ebroecker/canmatrix #324: Add lxml-stubs for xml type checks?

Funth0mas, 2019-03-07:
The `lxml` library have no annotations by default. But here is a [lxml-stubs](https://github.com/JelleZijlstra/lxml-stubs) project providing them. As the `arxml` Module heavily uses lxml, it might be usefull to include that annotations stubs so that mypy understands the context. 

@altendky  do you think it is a good idea? The project has no pip package, one has to clone it and add to own project.

Funth0mas, 2019-04-05:
After refactoring the arxml I see that we really need to do it somehow...

closed 2019-04-11 by Funth0mas

altendky, 2019-07-03:
Sorry I missed this.  :[  I have been pretty booked up and haven't been properly monitoring my email.  It doesn't seem like `canmatrix` should be providing generated stubs for other projects, no.  If there's an existing package it could just be installed as part of a dev package then?  I never did get the `requirements.txt` setup well for `canmatrix` but you can specify `git+https://github.com/JelleZiljlstra/lxml-stubs` in there (or maybe in a new `-dev.txt`?  or maybe this is mostly needed for test?).  Or if they really are generated then just generate them when needed, perhaps.