ebroecker/canmatrix #344: cannot import name 'deepcopy' from 'copy'

kamarajanis, 2019-04-03:
which library to install for this

ebroecker, 2019-04-03:
You don't need any.
Which Version of Canmatrix did you try?
Could you try current development branch?

Funth0mas, 2019-04-03:
I'd rather ask what exactly do you do? Such error happens if you try to run the CLI module as a script, not as a module. It must be ran as `python -m canmatrix.convert` or newly `python -m canmatrix.cli.convert`

ebroecker, 2019-04-26:
@kamarajanis 
is this issue still present? Do you have further infos for us?

kamarajanis, 2019-04-29:
I used latest version installed from pypi index

kamarajanis, 2019-04-29:
C:\Python37\lib\runpy.py:125: RuntimeWarning: 'canmatrix.convert' found in sys.modules after import of package 'canmatrix', but prior to execution of 'canmatrix.convert'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
ERROR - formats - This file format is not supported for writing

i get this error @Funth0mas  when  executing python -m canmatrix.convert

kamarajanis, 2019-04-29:
Thank you very much i understand how to run and the issue was solved

closed 2019-04-29 by kamarajanis

Funth0mas, 2019-04-29:
...and I hope the RuntimeWarning is already fixed in the development branch by restructuring the project.