ebroecker/canmatrix #345: reading of arxml fails in arGetPath because of type-mismatch

tiolan, 2019-04-04:
I have a minimal software, reading an arxml file.
Canmatrix was installed using pip resp. pip3.

>import canmatrix
>db = canmatrix.formats.load('/path/to/.arxml', "arxml")

I get the error message (x-out some path)

> Traceback (most recent call last):
>  File "arxmlreader.py", line 9, in <module>
>    db = canmatrix.formats.load('/xxxxxxx/.arxml', "arxml")
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/formats.py", line 74, in load
>    dbs = moduleInstance.load(fileObject, **options)
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/arxml.py", line 1634, in load
>    portRef = arGetChildren(sigTrig, "I-SIGNAL-PORT", searchPoint, ns)
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/arxml.py", line 942, in  arGetChildren
>    rettemp.append(arGetPath(arTranslationTable, ret.text))
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/arxml.py", line 910, in arGetPath
>    ptr = ptr.getChild(p)
>AttributeError: 'lxml.etree._Element' object has no attribute 'getChild'

For debugging I did in arGetPath(...):
>           if ptr is not None:
>                print(type(ptr))
>               print(type(p))
>               ptr = ptr.getChild(p)
>            else:
>                return None

which returns
><class 'lxml.etree._Element'>
<class 'str'>

Hence, ptr=ardict is not instance of arTree.

Python Version: Python 3.4.3 (default, Nov 12 2018, 22:25:49)

Could you please kindly check if this is a bug?

ebroecker, 2019-04-04:
please try:
```python
db = canmatrix.formats.load('/xxxxxxx/.arxml', "arxml", arxmlUseXpath=False)
```

tiolan, 2019-04-04:
No, this did not help
>Traceback (most recent call last):
>  File "arxmlreader.py", line 9, in <module>
>    db = canmatrix.formats.load('/xxx/.arxml', "arxml", arxmlUseXpath=False)
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/formats.py", line 74, in load
>    dbs = moduleInstance.load(fileObject, **options)
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/arxml.py", line 1662, in load
>    frameData = frame.encode(sig_value_hash)
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/canmatrix.py", line 946, in encode
>    return self.signals_to_bytes(data)
>  File "/xxx/virtenv/local/lib/python3.4/site-packages/canmatrix/canmatrix.py", line 892, in signals_to_bytes
>    least = self.size * 8 - signal.startBit
>TypeError: unsupported operand type(s) for -: 'int' and 'str'

ebroecker, 2019-04-04:
well - this is a new issue ...

do you have "multiplexed" frames in your ARXML?
There could be a bug in the arxml-importer which could lead to this issue.

tiolan, 2019-04-04:
Honestly, I am not really familiar with the structure of arxml files.
Could you please give me a hint how to find out if there are "multiplexed" frames in?

ebroecker, 2019-04-04:
... well could you try current developement branch?

tiolan, 2019-04-05:
with the dev branch I got:
>code:
>import canmatrix.formats.arxml
>db = canmatrix.formats.arxml.load('/xxx.arxml', "arxml", arxmlUseXpath=False)

>Traceback (most recent call last):
>  File "../src/arxmlreader.py", line 8, in <module>
>    db = canmatrix.formats.load('/xxx/.arxml', "arxml", arxmlUseXpath=False)
>  File "/src/canmatrix/canmatrix-development/src/canmatrix/formats/__init__.py", line 91, in load
>    dbs = module_instance.load(file_object, **options)  # type: ignore
>  File "/src/canmatrix/canmatrix-development/src/canmatrix/formats/arxml.py", line 1717, in load
>    ecu = process_ecu(node, db, searchPoint, multiplexTranslation, ns)
>  File "/src/canmatrix/canmatrix-development/src/canmatrix/formats/arxml.py", line 1500, in process_ecu
>    bu = ecu(arGetName(ecu, ns))
>TypeError: 'lxml.etree._Element' object is not callable

>code:
>import canmatrix.formats.arxml
>db = canmatrix.formats.arxml.load('/xxx.arxml') 

>Traceback (most recent call last):
>  File "../src/arxmlreader.py", line 8, in <module>
>    db = canmatrix.formats.arxml.load('/xxx.arxml')
>  File "/src/canmatrix/canmatrix-development/src/canmatrix/formats/arxml.py", line 1717, in load
>    ecu = process_ecu(node, db, searchPoint, multiplexTranslation, ns)
>  File "/src/canmatrix/canmatrix-development/src/canmatrix/formats/arxml.py", line 1500, in process_ecu
>    bu = ecu(arGetName(ecu, ns))
>TypeError: 'lxml.etree._Element' object is not callable

ebroecker, 2019-04-05:
ok, this helps.
You are reading an ECU extract. This feature is quite new...
Gonna have a look on it

Funth0mas, 2019-04-05:
Could you pls try the branch with open PR? It is checked using static typing and such error shouldn't happen anymore

tiolan, 2019-04-05:
Yes, that seems to work. Thanks.
I will continue with my implementation.

commit 886109e3 (2019-04-05): Annotate and cleanup arxml (#346)

very nice rework of horrible arxml module.
by the way this pr fixes #345
closed 2019-04-05 by ebroecker