ebroecker/canmatrix #352: Feature: Support special OBD2 DBC structure

MatinF, 2019-04-17:
Hi canmatrix team,

A suggestion for a potential feature:

**Context**
Many users frequently need to convert raw CAN data containing OBD2 response information, as per the standard defined for OBD2 Mode 01 PIDs: https://en.wikipedia.org/wiki/OBD-II_PIDs

To convert this data, the obvious solution would be a DBC file. However, the challenge is that the "trigger ID" for OBD2 PIDs is not only the CAN ID - but also depends on the first 3 data bytes. 

This means that OBD2 conversion via DBC would require special handling - a bit like the recent addition of the J1939 PGN special case in canmatrix.

**Example**
CAN Frame: 01T195108397;0;7e8;03410d00aaaaaaaa

Here, the 11 bit CAN ID is **7E8** (HEX) - which is the same for all OBD2 messages.

The first 3 data bytes are **03410d**. This can be considered a unique identifier for Vehicle Speed as follows: 
- 03 reflects that the following 3 bytes contain valid data
- 41 reflects that this is a response message, rather than a polling message for Vehicle Speed
- 0D reflects the Vehicle Speed parameter ID

Finally, the 4th data byte **00** is the vehicle speed data byte. In this case, the scaling is simply a factor of 1 and an offset of 0, meaning that 00 reflects 0 km/h.

**Implementation**
Similar to J1939 being identifiable via a protocol meta data tag in the DBC, an OBD2 file could be recognized in the same way - see attached example file.

[obd2.zip](https://github.com/ebroecker/canmatrix/files/3089907/obd2.zip)

In this case, the OBD2 protocol handling would mean that the "trigger ID" would not just be the CAN ID of 7E8 - but it would also include the first 3 data bytes, which could then be contained as part of the Signal name information (see the DBC for an example).

Similar to the J1939 conversion, our interest in this is indirect via the asammdf tool that uses canmatrix as a dependency - but I think this would be a relevant addition for many users as OBD2 is one of the most commonly used protocols.

I would be interested in hearing your thoughts on this - and whether this has been discussed previously. If you think it makes sense to add, I'm happy to discuss further and provide data samples or more extensive OBD2 examples.

Thanks again for your great work,
Martin

ebroecker, 2019-04-18:
Hi @MatinF ,

I had a look on it. I think this can be done by using "complex multiplexed" frames.
These frames can contain more than one multiplexer.
I'll try following:
* First Multiplexer: service
* second Multiplexer PID

The drawback is, canmatrix currently does not yet support decoding "complex multiplexed frames".
But with this use-case I'm gonna try to implement.

Another drawback is, that with this attempt only "single frames" (SF) can be supported, ISO-TP MF cannot really be supported by a single dbc.

The Attachment is a starting-point of a obd2-dbc ...
[obd2.zip](https://github.com/ebroecker/canmatrix/files/3093179/obd2.zip)

MatinF, 2019-04-18:
Hi Eduard, 

I think the multiplexing sounds like a very good idea, I hadn't thought about using it that way. Admittedly, multiplexing is not my strongest side, so it's a bit difficult for me to evaluate the OBD2 DBC sample you sent me - however, if I'm understanding the logic correct, then it looks like the right handling. 

I've tried to attach a data set in CSV format with actual OBD2 data from a car. The data set contains e.g. data for Vehicle Speed (PID 0D) and RPM (PID 0C), so it might be helpful in evaluating if the DBC performs the right way.

[OBD2_Data.zip](https://github.com/ebroecker/canmatrix/files/3094106/OBD2_Data.zip)

In regards to the multi frame challenge, I don't believe the standard OBD2 parameters require multiframe packets, though perhaps some proprietary PIDs do. I think that would be an acceptable drawback for this purpose.

One additional "tricky" part of OBD2 PIDs relate to some of them requiring two scaling parameters to be applied. You can see for example the PID 0C (engine RPM). Contrary to Vehicle Speed, RPM utilizes two data bytes (byte 3 and 4, counting from 0). These need to be scaled as below:

RPM = 64 * A + 0.25 * B 

Here, A and B are Byte 3 and 4 in decimal form. The challenge is that to my knowledge, DBC files do not support the use of two scaling parameters for a single signal. But maybe you have an idea of how this could be handled? One "less ideal" option would be to capture this through two signals and the end user would then have to add these together - but it's not a perfect solution.

It may be that there's another database format that is better suited for that level of complexity - I just thought it would be beautiful if the DBC format could somehow handle this as well, similar to how you propose a great solution with the multiplexing. 

Thanks for your inputs on this Eduard, it's a topic that's been nagging us for some time.

Best,
Martin

MatinF, 2019-04-19:
By the way in case the example with RPM is unclear, I'm happy to provide a more explicit example. Let me know.

Martin

MatinF, 2019-04-23:
One other thought: In case the RPM challenge cannot be supported within the *.DBC format, would one of the other database formats you're able to import work for this purpose? (i.e. multiple independent scaling factors)

We were considering e.g. the yaml format. Our thinking is that we'd rather utilize something that is standard/semi-standard to build up an OBD2 database, rather than do our own custom implementation. 

Happy to hear your thoughts on this Eduard,
Martin

ebroecker, 2019-04-25:
Hi @MatinF 

I had success decoding complex multiplexing messages now.
The code is very basic for now and have to be reworked to get integrated into canmatrix. But this will be done in the next days. I updated the dbc to be able to decode your complete example data.

the RPM-Challenge is more easy to solve than first thought:
```
RPM = 64 * A + 0.25 * B

RPM = (256 * A + B) /4

RPM = (A<<8 + B) /4
```
which means actual that RPM is a 16-bit-variable (motorola coded) and if modeled this way in dbc it does not need any special handling in decoding.

MatinF, 2019-04-25:
Hi Eduard,

That's an amazing observation - we've worked for a long time with this and never thought about it that way. Really a clever fix to this - exactly what I was hoping would exist. 

Could you share the DBC sample incl. RPM and this solution? Then I'd try adding a few to check that I interpret the methodology correctly.

When the code is updated, let me know - then I'll check-in with Daniel from asammdf to hear if it's possible to integrate there to facilitate OBD2 conversion of MDF data via DBC (which would be awesome).

Thanks for your great help on this!

Best,
Martin

ebroecker, 2019-04-25:
the code is far from complete.
the dbc is also far from complete. its minimal for decoding your example-data.

as said already, it's not integrated already - not even in branch - so just a prove of concept.

https://gist.github.com/ebroecker/c1ab21378f515ce9dd0e99e181fe0b79

[dbc_and_asc.zip](https://github.com/ebroecker/canmatrix/files/3116857/dbc_and_asc.zip)

MatinF, 2019-04-25:
Hi Eduard, fully understand - thanks for sharing the minimal example.

best,
Martin

ebroecker, 2019-04-26:
started branch `complex_multiplexing`
decoding should be possible with this branch.

ebroecker, 2019-04-30:
@MatinF 
did you have a chance to look at the snippets?
Does it fullfill your needs?

MatinF, 2019-04-30:
Hi Eduard, sorry for not replying yet - I've checked in with Daniel on this; he'll be back in a week and then I think he'll give it a look. I'll try to get some MDF log file samples with OBD2 data and provide this together with your DBC sample and code to see if it's possible to convert as intended. 

Will keep you posted on this :-)

danielhrisca, 2019-05-02:
> did you have a chance to look at the snippets?
> Does it fullfill your needs?

@ebroecker 
Your code was a valuable guide for handling multiplex signals. I was able to implement this function (tailored for my library and numpy) https://github.com/danielhrisca/asammdf/blob/development/asammdf/blocks/utils.py#L1314

ebroecker, 2019-05-02:
@danielhrisca 
you are using canmatrix anyway?

So just my question:
Why do you need to recreate the mux decoding at your side? 

I also implement this here in Canmatrix (branch complex_multiplexing)...

danielhrisca, 2019-05-02:
The speed is the main issue. Consider the following snippet, the signals extracted using Frame.decode still need to be aggregated 

```python

# without canmatrix decode
extracted_signals = extract_mux(payload, message, msg_id, bus, t)

# using canmtrix
x = payload.tobytes()
for i in range(len(x) // 8):
    extracted_signals = message.decode(x[i*8:(i+1)*8])
```

The timings are 6us without and 12ms with canmatrix decode

ebroecker, 2019-05-02:
OK, thanks for this hint. 
I had not in mind, that decoding speed an issue for now.
Gonna have a look on it: #362

maybe I lend some code from you to decode frames...

MatinF, 2019-05-02:
Hi Eduard, just wanted to say thanks for the good inputs on the whole OBD2 conversion methodology. With the implementation in asammdf, it's now possible to directly convert OBD2 PIDs via the DBC file, which is really great! Note that we had to make a small change in the DBC, shifting the positions of 'response' and 'service' - see attached adjusted DBC.

I'll try to add all PIDs in the coming period to create an OBD2 DBC that covers all the main parameters in Mode 01 of OBD2. Will also try creating some longer data samples for testing purposes.

Thanks again for all the help!

[obd2.zip](https://github.com/ebroecker/canmatrix/files/3137885/obd2.zip)

danielhrisca, 2019-05-02:
> I had not in mind, that decoding speed an issue for now.

I'm sure it's not if you do this live, while logging real traffic. For offline processing it shows.

> maybe I lend some code from you to decode frames...

Of course, it's totally up to you. I don't mind

ebroecker, 2019-05-02:
@MatinF  @danielhrisca 
thanks for you Feedback.

@MatinF I also changed the response and service in one of my dbcs. Maybe I gave you the old one...
With some work I could also create a ISOTP-Decoder (like the J1939Decoder) which also can work with multi frame answers...

danielhrisca, 2019-05-02:
@ebroecker 
if looks like sometimes _\_get_sub_multiplexer_ returns _None_ even though _\_has_sub_multiplexer_ return _True_
```
Traceback (most recent call last):
    extracted_signals = message.decode(x[i*8:(i+1)*8])
  File "...\_venv\lib\site-packages\canmatrix\canmatrix.py", line 1283, in decode
    decoded)
TypeError: cannot unpack non-iterable NoneType object
```

MatinF, 2019-05-02:
> @MatinF @danielhrisca
> thanks for you Feedback.
> 
> @MatinF I also changed the response and service in one of my dbcs. Maybe I gave you the old one...
> With some work I could also create a ISOTP-Decoder (like the J1939Decoder) which also can work with multi frame answers...

I think handling of multiframe packets would be very attractive to some of the J1939 users in particular. However, I do not yet have an MDF sample with multi frame packets for the purpose of testing. Will try and get some files when possible on this

ebroecker, 2019-05-02:
> 
> 
> @ebroecker
> if looks like sometimes __get_sub_multiplexer_ returns _None_ even though __has_sub_multiplexer_ return _True_
> 
> ```
> Traceback (most recent call last):
>     extracted_signals = message.decode(x[i*8:(i+1)*8])
>   File "...\_venv\lib\site-packages\canmatrix\canmatrix.py", line 1283, in decode
>     decoded)
> TypeError: cannot unpack non-iterable NoneType object
> ```
> 
> This is with the latest _complex_multiplexing_ code

I could imagine the multiplexer is not in range `signal.multiplexer_value_in_range(parent_multiplexer_value)`:
https://github.com/ebroecker/canmatrix/blob/05c620ac8d27fea1928c39b0d518f546808e2f43/src/canmatrix/canmatrix.py#L1239

MatinF, 2019-05-17:
Hi Eduard, would it be possible to merge the extended multiplexing functionality to the canmatrix master?

Best,
Martin

ebroecker, 2019-05-17:
its already merged to development:
https://github.com/ebroecker/canmatrix/pull/356

to master usually only release are merged.
is it important for you, that it's merged to master?

MatinF, 2019-05-17:
We use it indirectly via asammdf. Currently, we are able to do OBD2/J1939 conversion if we install the development branch of canmatrix, whereas the master 0.7 does not work. So I guess in practice it might be both the complex multiplexing and the J1939 decoding that is currently part of development which we'd like to have as part of the master as it simplifies the installation of asammdf for our users.

Alternatively is it then possible to make asammdf require the development rather than master version of canmatrix - so that this is installed by default when users install asammdf?

MatinF, 2019-05-17:
Currently we solve this by running the below:
```
pip install asammdf
pip install git+https://github.com/ebroecker/canmatrix.git@development
```
But would be cleaner to be able to leave out the 2nd line if possible. But of course understand if this has to await the release - we hope it's possible to include in the next one the OBD2 and J1939 support.

ebroecker, 2019-05-17:
well ... I didn't panned a release yet. 
[There is also an issue (#291) about release frequency] 

I'll think about a release...

MatinF, 2019-05-17:
It's of course up to you Eduard, we're already grateful for the support on both J1939 and OBD2, it means a ton for our work

danielhrisca, 2019-05-17:
> well ... I didn't panned a release yet.
> [There is also an issue (#291) about release frequency]
> 
> I'll think about a release...

being able to pip install makes things a lot easier for the end users

ebroecker, 2019-05-17:
@MatinF and @danielhrisca 

0.8 released ...

MatinF, 2019-05-17:
That's aweosme Eduard, I tested it and it looks like the new master works similarly to the development branch, thanks a lot! Closing the issue here.

closed 2019-05-17 by MatinF

vmsuk, 2023-11-12:
Hi,
Is there any place online to get/buy only following 4 OEM Extended PID?
1- Ododmeter
2- Fuel Level
3- DTC
4- VIN 

Thanks