ebroecker/canmatrix #358: error with line no:31 like many lines means is the dbc encrypted ??

kamarajanis, 2019-04-29:
while converting dbc to xlsx i get a error 
saying that error with the particular line and also printing the line 
is this encrypted dbc 
or not able to convert that line 
any another wat to do that??

ebroecker, 2019-04-29:
could you post the line?

kamarajanis, 2019-04-29:
ok 
C:\Python37\lib\runpy.py:125: RuntimeWarning: 'canmatrix.convert' found in sys.modules after import of package 'canmatrix', but prior to execution of 'canmatrix.convert'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
error with line no: 31
b' SG_ AccSts : 62|3@0+ (1.0, 0.0) [0.0|0.0] "" VDDM\n'
error with line no: 33
b' SG_ AdjSpdLimnWarn : 59|2@0+ (1.0, 0.0) [0.0|0.0] "" VDDM\n'
error with line no: 35

kamarajanis, 2019-04-29:
like this many line are not decoded

kamarajanis, 2019-04-29:
And also it decoded the Signals  having value signal bit length 1 
how to decode all signals ??

ebroecker, 2019-04-29:
@kamarajanis 

Thank you very much for your bug report.

Please try branch `iss358`:

`pip install git+https://github.com/ebroecker/canmatrix@iss358#egg=canmatrix`

Thanks

kamarajanis, 2019-04-29:
thank you 
but And also it decoded the Signals having value signal bit length 1
you have given FollowUps.signalComment is 1 as try block
how to change to any value and decode all signals ??

ebroecker, 2019-04-29:
Hi @kamarajanis 
not sure what you mean.
Usually dbc-import should be fairly complete.
What is the issue like?

the `FollowUps.signalComment`  is only needed for multi-line comments. It should not bother the frame/signal reading functionality.

kamarajanis, 2019-04-29:
It only extracted the signals that have signal  length bit 1 and printed in that xlsx file 
Not all signals
And also what is meaning of each and every 
Line in the dbc when you open as txt format 
Is there any documentation to that study

ebroecker, 2019-04-29:
do you have any further error messages from canmatrix?
are you sure there are other signals?
there is no signal-length check anywhere in canmatrix dbc import.

you can find some infos about dbc here: http://socialledge.com/sjsu/index.php/DBC_Format

kamarajanis, 2019-04-29:
NO ok thank you for that reference

ebroecker, 2019-04-30:
@kamarajanis 
this issue is fixed.
Just open a new issue, if any other problem occures

closed 2019-04-30 by ebroecker