ebroecker/canmatrix #361: Converting arxml to dbc

shubhamthakare, 2019-05-02:
Hello I have Canmatrix

Please help me how to convert an arxml file to dbc

Thanks in advance

ebroecker, 2019-05-02:
https://canmatrix.readthedocs.io/en/latest/cli.html#conversion-of-file-formats

shubhamthakare, 2019-05-02:
Should i directly run the script canconvert.py?

shubhamthakare, 2019-05-02:
script is getiing executed ,but output dbc is not getting created

shubhamthakare, 2019-05-02:
Please help

ebroecker, 2019-05-02:
could you PLEASE provide more information.

* what did you exactly do?
* what are the errors/logs?

shubhamthakare, 2019-05-02:
i directly executed the script canconvert.py myarxml.arxml mydbc.dbc

shubhamthakare, 2019-05-02:
PFA

shubhamthakare, 2019-05-02:
I tried as below

"W:\5. Wrk\SWD\Shubham P\canmatrix-development\src\canmatrix\convert.py "E3_1_2_Premium_V12.01.20F_AR430_20190405_BMC_HV_SEn.arxml j1939.dbc

ebroecker, 2019-05-02:
any errors?
did you try `W:\5. Wrk\SWD\Shubham P\canmatrix-development\examples\convert.py`

did you try option `-vv`

do you really have no output? While converting from `ARXML` there will be no j1939.dbc but some *j1939*.dbc

shubhamthakare, 2019-05-02:
No i didnt get any output

shubhamthakare, 2019-05-02:
i tried W:\5. Wrk\SWD\Shubham P\canmatrix-development\examples\convert.py this too
where the dbc will be geenerated ?
I am unable to locate in my directory

ebroecker, 2019-05-02:
even no console output with option `-v` or `-vv` (`canconvert -vv some.arxml some.dbc`)

shubhamthakare, 2019-05-02:
getting the below error 

C:\Users\PatilShu>"W:\5. Wrk\SWD\Shubham P\canmatrix-development\examples\convert.py -vv E3_1_2_Premium_V12.01.20F_AR430_20190405_BMC_HV_SEn.arxml test.dbc                                                                                  '"W:\5. Wrk\SWD\Shubham P\canmatrix-development\examples\convert.py -vv E3_1_2_Premium_V12.01.20F_AR430_20190405_BMC_HV_SEn.arxml test.dbc' is not recognized as an internal or external command,

shubhamthakare, 2019-05-02:
'"W:\5. Wrk\SWD\Shubham P\canmatrix-development\examples\convert.py -vv"' is not recognized as an internal or external command,                                                                                                              operable program or batch file.

ebroecker, 2019-05-02:
you need to do this on command line

shubhamthakare, 2019-05-02:
one dbc is created ..but it is empty .and when i enter the command one popup is coming "syntax error"
Unable to open the file

The file "convert.py -vv E3_1_2_Premium_V12.01.20F_AR430_20190405_BMC_HV_SEn.arxml test.dbc" is generated and it is empty

shubhamthakare, 2019-05-06:
hii @ebroecker please help me 
i am unable to conveet arxml to dbc

ebroecker, 2019-05-06:
You have to start canconvert in a command shell like `cmd.exe`.

Than you will get `canconvert -vv` running and you will get log messages.
I can only help if you provide these log messages.

An other option would be to provide your ARXML-File to me (maybe by pn).

shubhamthakare, 2019-05-07:
I am starting canconvert.py in command shell only but i am not getting any log messages 
Sorry I can't share arxml as i cant share it through my company due to its security policy

ebroecker, 2019-05-07:
with option `-vv` you always should have an output.
Please try to convert the `test.arxml` which is `canmatrix/test` folder:

```cmd.exe
C:\.....\canmatrix\test>python -m canmatrix.cli.convert -vv test.arxml aa.dbc
INFO - convert - Importing test.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done
```

the command above should generate a dbc: `aa_CAN.dbc`

shubhamthakare, 2019-05-07:
No its not creating the dbc 
Do i need to keep arxml file in the same folder where the convert script is there ?

shubhamthakare, 2019-05-07:
[image]

ebroecker, 2019-05-07:
`python -m canmatrix.cli.convert..`
not 
`src\canmatrix\cli\convert.py -m ...`

shubhamthakare, 2019-05-07:
[image]

ebroecker, 2019-05-07:
just `python` not `some\path\python`

https://docs.python.org/3/faq/windows.html

btw. what version of canmatrix did you install?
Did you install at all?

did you also
`pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[arxml]`?

https://canmatrix.readthedocs.io/en/latest/installation.html

shubhamthakare, 2019-05-09:
[image]

shubhamthakare, 2019-05-09:
i have Python 2.7 installed unable to install canmatrix

shubhamthakare, 2019-05-09:
[image]

Funth0mas, 2019-06-12:
Let me summarize a little bit:
- canmatrix is command line tool and therefore requires slightly advanced computer skills
- prefer python 3.7 over 2.7 for which even the pip will drop support soon
- if you install python on you own, tick "add python.exe to PATH" during install
- if you can't, you have to write full path to `python.exe`, not just `>python -m something`
- carefully read and follow instructions
- don't use directories with spaces like `5. Wrk` or `Shubham P` unless you're skilled enough to cope with them. Use underscore '_' instead (5_Wrk etc.)
- carefully read and follow instructions
- prefer installing canmatrix using pip (`pip install canmatrix` or `C:\path_to_python\python.exe -m pip install canmatrix`) https://pypi.org/project/canmatrix/ over clonning git, (if you don't need development branch and if you aren't behind evil corporate NTLM proxy, this is yet another story)
- carefully read and follow instructions
- carefully read and follow instructions

@ebroecker I think the ticket can be closed, right?

closed 2019-06-25 by ebroecker

Ram2763, 2024-12-27:
Facing below error while workign on arxml to .dbc converion, 
usig this command : python -m canmatrix.cli.convert -vv CHG00.arxml CHG00.dbc

Command window tarce:
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::HCHGILM_dms65_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::ECAN_HTD_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::OVL_HTD_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::SMRW_HTD_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::EVL_HTD_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::EPCHG_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::HM_HTD_dms65_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::HEL_HTD_dms65_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::HEH_HTD_dms65_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::EDCDC_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::VB_HTD_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::HWT_HTD_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
WARNING - dbc - Signal EHV1E09_var10_oCANFD_BT_Bus::CHGCCTST_dms00_oEHV1E09_var10_oCANFD_BT_Bus name exceeds 32 characters, consider updating the signal name within the character limit(Max 32 characters)
Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python39\lib\site-packages\canmatrix\cli\convert.py", line 160, in <module>
    sys.exit(cli_convert())
  File "C:\Program Files\Python39\lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "C:\Program Files\Python39\lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\Program Files\Python39\lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Program Files\Python39\lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\Program Files\Python39\lib\site-packages\canmatrix\cli\convert.py", line 155, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Program Files\Python39\lib\site-packages\canmatrix\convert.py", line 351, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "C:\Program Files\Python39\lib\site-packages\canmatrix\formats\__init__.py", line 129, in dumpp
    dump(db, file_object, export_type, **options)
  File "C:\Program Files\Python39\lib\site-packages\canmatrix\formats\__init__.py", line 104, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "C:\Program Files\Python39\lib\site-packages\canmatrix\formats\dbc.py", line 327, in dump
    f.write(signal_line.encode(dbc_export_encoding, ignore_encoding_errors))
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2103' in position 67: ordinal not in range(256)