ebroecker/canmatrix #363: converting arxml 'CAN-FD' to dbc 'CAN-FD'

ilazrak, 2019-05-13:
hello, 

i want to convert an arxml 'CAN-FD' (where Signals are no longer attached directly to frames, but to PDUs, and each frame contains many PDUs) to dbc where i can see all the PDUs and their relative signals.

I tried 'canconvert' but i see that frames with many PDUs are not supported i got just signals within the first PDU not all of them.
There is a way i can convert correctly this kind of arxml with canmatrix ?

Thanks in advance :)

ebroecker, 2019-05-13:
Hi @ilazrak 

Thanks for your Issue-Report.

could you provide some sample `ARXML` data? Maybe some dummy-data?

Does one `PDU-TO-FRAME-MAPPING` contain multiple `PDU-REF` elements (pointing to `SIGNAL-I-PDU`)?

ilazrak, 2019-05-13:
attached some screenshots that show what the arxml looks like. 

are the screenshots sufficient to understand the arxml data?

[image]
[image]
[image]
[image]
[image]

thank you for your reply

ebroecker, 2019-05-14:
Thanks, I'll have a lokk on it.

did you try `--arxmlIgnoreClusterInfo` already?

ebroecker, 2019-05-14:
@ilazrak 

could you provide the node `/Communication/PDUs/PDU2`?

[image]

ilazrak, 2019-05-15:
@ebroecker 

[image]
[image]

did i sent the right node ? 

if not i'll try to create some dummy-data

wuyukk, 2019-05-16:
Hello,
Does this tool support convert dbc file to excel file ?
I tried, but failed:

INFO - convert -
INFO - convert - 299 Frames found
ERROR - __init__ - This file format is not supported for writing

ebroecker, 2019-05-16:
> 
> 
> Hello,
> Does this tool support convert dbc file to excel file ?
> I tried, but failed:
> 
> INFO - convert -
> INFO - convert - 299 Frames found
> ERROR - **init** - This file format is not supported for writing

ebroecker, 2019-05-16:
@ilazrak 
could you try branch iss363?

Thanks

ilazrak, 2019-05-21:
@ebroecker 
sorry for the delay 
i tried branch iss363, i got the same frames using the development branch, nothing changed.

Thanks

ebroecker, 2019-05-21:
@ilazrak 
I tried to create an ARXML myself: 
https://github.com/ebroecker/canmatrix/blob/ad922bcd819d6a8ee733ea1d7c5ee436972a72d1/src/canmatrix/tests/test_arxml.py#L19-L129

but this seems not to be right.
could you review this test-ARXML?
could you try the conversion with `-vv` and provide the output?

Or - maybe you could stip down you ARXML to one Frame, where you also rename the signals so that I could use a real live example? This would help.

ilazrak, 2019-06-06:
[arxml_Fd.zip](https://github.com/ebroecker/canmatrix/files/3261478/arxml_Fd.zip)
@ebroecker 

attached one Frame similar to my arxml data 

thanks

ebroecker, 2019-06-06:
@ilazrak 
thanks a lot, this will help!

I'm gonna work on it the next days...

ilazrak, 2019-06-06:
@ebroecker 
[ARXML.zip](https://github.com/ebroecker/canmatrix/files/3262152/ARXML.zip)

attached different ways to see a Frame in arxml

ebroecker, 2019-06-08:
@iartez 

could you have another look at branch iss363?
Thanks again for providing the test files!

ilazrak, 2019-06-11:
@ebroecker 
i tried branch iss363, i got empty containers after conversion.
did you successfully convert ARXMLtest.arxml ?

ilazrak, 2019-06-11:
@ebroecker 
im sorry, indeed it works correctly with branch iss363 

is there a way to get a dbc where signals are defined inside each PDU instead of frame ?

ebroecker, 2019-06-24:
maybe we could simulate with signal-groups, would this help?

ilazrak, 2019-06-25:
@ebroecker 
if you mean by signal-groups : "PDU type = I-signal-I-PDU". 
Yes this would help a lot !

Thanks,

ebroecker, 2019-07-15:
I updated branch 'iss363' just now.

could you please look to the generated dbc if it fulfills your needs?

I added a signal-group for each `container-i-pdu`

ebroecker, 2019-08-13:
to be continued in #385

closed 2019-08-13 by ebroecker