ebroecker/canmatrix #364: pyinstaller freez

danielhrisca, 2019-05-16:
For [asammdf](https://github.com/danielhrisca/asammdf) we use [appveyor](https://ci.appveyor.com/project/danielhrisca/asammdf/history) to generate stand-alone builds for the GUI application, using PyInstaller.

If the GUI is ran as a Python script there is no problem. Everything works without errors. 

However the stand-alone app freezez when the canmatrix functions are called (loading a DBC and extarcting CAN signals from raw CAN messages). There is no error message in the console.

Did anyone have such problems? I suspect it has to do with the dynamic loading of module inside canmatrix

ebroecker, 2019-05-16:
what way to load do you use?

`canmatrix.formats.load`?

maybe you could try to use `canmatrix.formats.dbc.load` directly - this should circumvent the dynamic loader...

danielhrisca, 2019-05-16:
I wilk try this and get back to you. Thank you

altendky, 2019-05-16:
@danielhrisca because of the dynamic imports pyinstaller can't detect them, iirc.

https://github.com/altendky/st/blob/b09b0ea34e2680819794e67ad299ac8b48fdef20/installer/pyinstaller.spec#L43-L44

```python3
for format in canmatrix.formats.moduleList:
    hidden_imports.add('canmatrix.' + format)
```

It's good to get some file logging for errors as well with `sys.excepthook`.  I use different stuff that's more of a mess and should be changed in my actual applications, but here's what I've got in my example app right now.

https://github.com/altendky/basicpyqt5example/blob/08457cf4fa303e7f3cd6be341725ac9cf8ee3a62/src/basicpyqt5example/__main__.py#L54-L62

```python3
logger.addHandler(file_handler)

def excepthook(excType=None, excValue=None, tracebackobj=None):
    logger.critical(''.join(traceback.format_exception(
        etype=excType,
        value=excValue,
        tb=tracebackobj,
    )).strip())
```

altendky, 2019-05-16:
@danielhrisca, side note, now that I finally look at what your repo is for you've definitely caught my interest.  I like standards and I also want to add some plotting to my project at some point.  I've been wanting to play a bit with PyQtGraph, especially realtime stuff.

danielhrisca, 2019-05-17:
Thank you @altendky  I will try your suggestions.

PyQtGraph work reasonably well even for live plotting as long as you don't have many samples (in the asammdf GUI I compute min and max so that I only need to plot 2 samples/pixel)

danielhrisca, 2019-05-17:
To my shame, the appveyor script was acutally installing the 0.7 release instead of the development branch. Fixing this together with importing the _load_ functions from _dbc_ and _arxml_ has fixed the problem.

closed 2019-05-17 by danielhrisca