ebroecker/canmatrix #367: Issue in converting an excel file to dbc

Jayanthdj, 2019-05-31:
Could you please help as I am facing issue in converting an excel files to dbc. I am attaching the screenshot
First file error
[image]
 second file error
[image]

Funth0mas, 2019-06-11:
Could you please provide minimal example of such excel file?

yuren1986, 2019-06-15:
Hello Funth0mas,

 I also find the same issue report.

[image]

So can we use the canconvert change the .xlsx file to .dbc ?

Hope your feedback, thanks.

Funth0mas, 2019-06-15:
Hi @yuren1986, both issues are pretty unrelated. The first has "incompatible" header whereas your table contains spaces in the cell, where canmatrix expects CAN ID ("ID" in the header). It must either contain valid hexadecimal CAN ID ended by "xh" or empty cell (really empty, not containing spaces not visible by eyes)

You can find example table under https://github.com/Funth0mas/canmatrix/blob/development/test/test.xls and https://github.com/Funth0mas/canmatrix/blob/development/test/test.xlsx  I see possible discrepancy with plain non-extended can IDs

If it doesn't help you, let provide your table (simplified, with modified data if they're secret) for analysis.

yuren1986, 2019-06-16:
Hello Funth0mas,
Thanks for your helping. I had checked your suggestion the test.xlsx can be transformed to dbc file, but the dbc file can't open.

The error as following, i think the reason is caused by the send type. 
[image]

And I use the TEST file cmTemplate.xlsx change to the .dbc , and I found the CAN ID has been changed from 123h to 12h.
[image]

So maybe the data information has been changed wrong by data type between transformation.]

Do you find the same issue also?

ebroecker, 2019-06-16:
Looks like the define ist Missing. I'll have a Look on it

Funth0mas, 2019-06-17:
There is indeed a discrepancy in parsing standard can ID (123h) versus extended (12345xh), in both cases are stripped last two letters, effectively leading to trimming 123h to 0x12.

yuren1986, 2019-06-17:
@Funth0mas @ebroecker I have got it.  How can we use this? Actually this extend mode and normal  mode doesnt be combined in one DBC. I will verify  with this on independently. Thanks for your analyze, hope you have a nice day.

Funth0mas, 2019-06-17:
If you need to convert standard can IDs, than simply try to write them as 123sh in the excel to get 0x123 in the dbc. It's just a hotfix until the bug will be fixed. You can safely mix standard and extended ids in single file.

ebroecker, 2019-06-24:
@yuren1986 

there are 2 issues here:
1. corrupt arbitration ids
2. the "GenMsgSendType"

for 1. I'll provide a fix soon, till than use the workarround presented aboven (from  @Funth0mas)
for 2: use `--deleteFrameAttributes=GenMsgSendType`
`python -m canmatrix.cli.convert test\test.xls foo.dbc --deleteFrameAttributes=GenMsgSendType`

2: only is a problem, if no send-type (e.g. cyclic) is defined anywhere in xls(x)

ebroecker, 2019-06-25:
@yuren1986 could you try branch `iss367` please?

should fix (1.)

Thanks

ebroecker, 2019-07-15:
(1. ) should be fixed with #377  (in development branch since 1ccdb5d39b175ea869c43d780a1cf01cbe8569f9)

closed 2019-08-19 by ebroecker

Ederwardlu, 2022-03-29:
There are something wrong as figure show when I try to convert xlsx to dbc 
[image]
.

commit 1b026d6a (2019-08-19): XLSX fixes  (#392)

XLSX: fixes #240 (signal init value in excel template)
XLSX: fixes #367 (2.)