ebroecker/canmatrix #368: Value type 'is_signed': converting arxml to dbc

ilazrak, 2019-06-07:


ilazrak, 2019-06-07:
@ebroecker 

'unsigned signals' in arxml become 'signed signals' after conversion to dbc,
which also affects the Min and Max values of the obtained signals...

        rawRange = 2 ** (self.size - (1 if self.**is_signed** else 0))
        return (
            factory(-rawRange if self.**is_signed** else 0),
            factory(rawRange - 1),

Funth0mas, 2019-06-11:
Could you please provide minimal example of such arxml file?

ilazrak, 2019-06-24:
[NewCluster_max_min.zip](https://github.com/ebroecker/canmatrix/files/3319776/NewCluster_max_min.zip)

@Funth0mas 
attached one Frame similar to my arxml data

thanks

Funth0mas, 2019-08-02:
Okay I accidentally reproduced the same issue with own ARXML :-) @ebroecker, there is some signed/unsigned guessing based on `BASE-TYPE` name. Instead, as described in https://www.autosar.org/fileadmin/user_upload/standards/adaptive/17-10/AUTOSAR_SWS_AdaptivePlatformTypes.pdf, we shall check for `<BASE-TYPE-ENCODING>` value:

 * `NONE` for unsigned integers
 * `2C` for signed integers "two's complement"
 * `IEEE754` for float32/float32
 * `BOOLEAN`

have you already some progress, or shall I try to fix it? I might find some time to do it.

ebroecker, 2019-08-02:
@Funth0mas 
You are probably right...
Well - you are welcome - if you find time - do fix this

Funth0mas, 2019-08-08:
@ilazrak please does the linear scale used in you example file make sense? There are
```
                      <COMPU-NUMERATOR>
                        <V>0</V>
                        <V>1</V>
                      </COMPU-NUMERATOR>
                      <COMPU-DENOMINATOR>
                        <V>0</V>
                        <V>1</V>
```
If I understand correctly that is a polynomial (0 + 1*x)/(0 + 1*x) = 1, that means constant one... unfortunately this also leads to divide-by-zero in canmatrix because it expects such function to be expressed simply as (1 + 0*x)/(1 + 0*x) => offset=1/1=1, scale=0/1=0.

ebroecker, 2019-08-13:
@iartez could you please try branch iss385?

Thanks

ebroecker, 2019-08-19:
fixed in development now.
Please reopen or open new issue if still not working as expected

closed 2019-08-19 by ebroecker

Prabhu6296, 2021-01-25:
> @ilazrak please does the linear scale used in you example file make sense? There are
>
> ```
>                       <COMPU-NUMERATOR>
>                         <V>0</V> [...]

Could you please elaborate and explain the calculation of offset and scale value? It's a little confusing due to 1 and 0 values. Thanks in advance :) hope you will take some time to rectify this minuscule query of mine!

Funth0mas, 2021-03-27:
@Prabhu6296  you can find more information in https://www.autosar.org/fileadmin/user_upload/standards/classic/4-2/AUTOSAR_TPS_SoftwareComponentTemplate.pdf , section "5.5.1.1    Category Values in the context of a CompuMethod". Than  search for `CompuNominatorDenominator`. 
The offset and scale is expressed os a 2nd order polynomial.