ebroecker/canmatrix #375: UnicodeEncodeError: 'latin-1' codec can't encode character '\u2026'

cbaciu, 2019-07-03:
Traceback (most recent call last):
  File "c:\python35\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python35\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python35\Scripts\canconvert.exe\__main__.py", line 9, in <module>
  File "c:\python35\lib\site-packages\canmatrix\cli\convert.py", line 237, in main
    canmatrix.convert.convert(infile, out_file_name, **cmdlineOptions.__dict__)
  File "c:\python35\lib\site-packages\canmatrix\convert.py", line 206, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "c:\python35\lib\site-packages\canmatrix\formats\__init__.py", line 132, in dumpp
    dump(db, file_object, export_type, **options)
  File "c:\python35\lib\site-packages\canmatrix\formats\__init__.py", line 107, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "c:\python35\lib\site-packages\canmatrix\formats\dbf.py", line 514, in dump
    f.write(out_str.encode(dbf_export_encoding))
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2026' in position 574161: ordinal not in range(256)

altendky, 2019-07-03:
It looks like you have an ellipsis in your data and need to pick an encoding which supports it.  Or, get rid of the ellipsis.  You should be able to specify `--dbcExportEncoding=utf-8` on the command line to use UTF-8 as the output encoding, for example.

@ebroecker, is iso-8859-1 really a good default as opposed to UTF-8?  Though I see latin-1 in the traceback...  I always forget if that's synonymous with iso-8859-1 or not.

ebroecker, 2019-07-16:
At least at my side `dbf ` and `dbc` tools want `latin-1` coded files. 
I have never tested if this depends on the local setting of the system.
Yes `iso-8859-1 == latin-1`

What about fixing this like:
```python
f.write(out_str.encode(dbf_export_encoding, 'ignore'))
```

altendky, 2019-07-16:
I would probably only do that as an optional thing.  Canmatrix could catch the above exception and include a message letting the user know that if they want to coerce and lose data they can pass `--ignore-encoding-errors` or somesuch.

Do the .dbc programs fail with UTF-8 input?  *sigh*  it's going to be awhile yet before we can really just use UTF-8, isn't it...

closed 2019-08-14 by ebroecker