ebroecker/canmatrix #376: Issue in dump

Abhi1gu, 2019-07-09:
canmatrix.formats.dbc.dump(db1, "testDBC.dbc")
Traceback (most recent call last):
  File "<pyshell#63>", line 1, in <module>
    canmatrix.formats.dbc.dump(db1, "testDBC.txt")
  File "C:\Users\4NONYM4U5\AppData\Local\Programs\Python\Python37-32\lib\site-packages\canmatrix\formats\dbc.py", line 163, in dump
    f.write(header.encode(dbcExportEncoding))
AttributeError: 'str' object has no attribute 'write'

ebroecker, 2019-07-10:
you need to use a file descriptor instead of a filename:
either use formats:
```python
canmatrix.formats.dumpp(db1,  "testDBC.txt", export_type="dbc")
```
or open file first:
```python
with open("testDBC.txt", "wb") as fd:
    canmatrix.formats.dbc.dump(db1, fd)
```

Abhi1gu, 2019-07-10:
Thanks for your feedback. That worked!

closed 2019-07-15 by ebroecker