ebroecker/canmatrix #381: recalc_J1939_id() changing id of all frames in version 0.8

FilipZZZ, 2019-08-05:
Hi,
I tried using version 0.8 for generating J1939 .dbc and unfortunately it doesn't work, probably due to the new ArbitrationId class.
When I call the recalc_J1939_id() so that the message IDs don't stay zero, it forces the last recalculated id (from a j1939 pgn) from one frame(https://github.com/ebroecker/canmatrix/files/3466457/createJ1939DbcRef.zip)
 upon all other frames. Therefore all the IDs in the generated .dbc file are the same.
When I went back to version 0.7 it solved the issue for me for now. 

Here is my edited example file for demonstration:
[createJ1939DbcRef.zip]

ebroecker, 2019-08-05:
@FilipZZZ 

Thanks for reporting this bug even with working example. 
This seems to be a wired issue ...

ebroecker, 2019-08-05:
found some hotfix:
https://github.com/ebroecker/canmatrix/tree/iss381

but not sure why it goes wrong without the 'deepcopy'

altendky, 2019-08-05:
Functions/methods that take multiple types can easily get confusing.  One path through `arbitration_id_converter()` gives you a new id instance, the other gives you an existing id instance.  Also, there's `attr.evolve()` though it is admittedly not a deep copy.  Immutable (`frozen=True`) instances can also be nice to help avoid these sorts of bugs.

ebroecker, 2019-08-06:
seems the default argument lead to the error. I changed to "0" - now it seems to do like expected.

https://github.com/ebroecker/canmatrix/blob/eaeab8e649d5a7ccfe4a8c22ca968d0d409aaccc/src/canmatrix/canmatrix.py#L723

ebroecker, 2019-08-13:
fix is in development branch now

closed 2019-08-13 by ebroecker