ebroecker/canmatrix #382: Support for SECURED-I-PDU

stefan325, 2019-08-11:
Hello ebroecker,

thanks for canmatrix. I have been using the examples/convert.yp script with a CAN FD Armxl (I am using your branch iss363 as described in issue https://github.com/ebroecker/canmatrix/issues/363), and it mostly works very well.
However, it looks that there presently is no support for SECURED-I-PDUs. Console output is "INFO - arxml - found secured pdu - no signal extraction possible: [...]".

I have a got an ARXML file with SECURED-I-PDUs and I am interested in that feature. Have you ever looked into SECURED-I-PDUs? It looks like the basic functionality could be added without too much trouble.
In the Pdus top-level package, there is a SECURED-I-PDU Element instead of the "normal" I-SIGNAL-I-PDU element. This elemnt points to a PAYLOAD-REF in the CommunicationClusters top-level package. The PAYLOAD-REF in turn, points to a regular I-SIGNAL-I-PDU, which contains the I-SIGNAL-I-PDU mappings.
What do you think?

[image]

ebroecker, 2019-08-12:
Hello @stefan325 

Thanks for your analysis.
I'd never looked into these 'secured' pdus. 

Gonna start implementing...

ebroecker, 2019-08-12:
... it seems to be easy to fix, could you try branch "iss382" ?

stefan325, 2019-08-12:
Hello @ebroecker,

I am impressed. It works, thank you!

Best,

Stefan

ebroecker, 2019-08-12:
@stefan325 

Thanks for your quick feedback.
Merged this small change into branch "iss363", will go mainline in the near future

closed 2019-08-12 by ebroecker