ebroecker/canmatrix #401: arxml with cascaded dictionary

mikesmitti, 2019-09-24:
Python version

('python=3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 21:26:53) [MSC v.1916 32 bit '
'(Intel)]')
'os=Windows-10-10.0.16299-SP0'
'numpy=1.16.4'
'asammdf=5.10.3'

MDF version
4.10

I use asammdf to open mf4 files.
There I found the behavior when using an arxml file belonging to the trace:
I get an exception when using the db-object, since there is no attribute "ecus"

File "C:__Programme\Python\Python_373\lib\site-packages\asammdf-5.10.3-py3.7.egg\asammdf\blocks\mdf_v4.py", line 574, in _read
board_units = set(bu.name for bu in db.ecus)
AttributeError: 'dict' object has no attribute 'ecus'

Description

I found  an arxml database with a cascaded dictionary. I mean, it got a dictionary "xyz" in it, which contains the dictionaries asammdf is looking for. This causes an exception, since the dictionaries looked for are not in the place they should be.

I changed the code from asammdf like this:
if db:
try:
board_units = set(bu.name for bu in db.boardUnits)
except AttributeError:
try:
board_units = set(bu.name for bu in db.ecus)
except:
board_units = set(bu.name for bu in db["DictnameHere"].ecus)

This works but is no solution for wider use.

It seems like the file is not the standard, due to this cascaded dictionaries.
But it would be good for canmatrix to be able to read this as well.

ebroecker, 2019-09-25:
Hi @mikesmitti 
 If I look at you work arround:
```python
if db:
  try:
    board_units = set(bu.name for bu in db.boardUnits)
  except AttributeError:
      try:
        board_units = set(bu.name for bu in db.ecus)
      except:
        board_units = set(bu.name for bu in db["DictnameHere"].ecus)
```

I see two issues:
1.` db.boardUnits` vs. `db.ecus`:  `boardUnits` is no attribute of canmatrix anymore. This is the old api. During cleanup we renamed to `ecus`. A canmatrix-object always has `ecus`.
2. `db.ecus` vs. `db["DictnameHere"].ecus`: loading an `ARXML`gives you a dictionary of canmatrix objects. What api do you use to load? Could - maybe - the flat-option (`loadp_flat`) helpful? This would give you always a canmatrix object (the first extracted bus definition only) and never a dictionary of canmatrix objects.
Or use something like `cluster = canmatrix.cancluster.CanCluster(dbs)` and afterwards you'll have the attribute `cluster.ecus`.

After reading you post again, I think, you are talking about issue 2. 
The background of these "cascading dictionaries" is that a format like `ARXML` can contain multiple bus definitions. While reading the file all busses are imported and will be returned like you found it. If you simply want some information about ecus, frames and signals no matter which bus they behave you should definitively have a look on `CanCluster`.

ebroecker, 2019-10-14:
@mikesmitti 
is this still open? Do you see things have to change? 
If so - please reopen

closed 2019-10-14 by ebroecker

mikesmitti, 2019-10-15:
I do not use it by myself, I use it via asammdf. But I will forward your repsonse.
Thank you and best regards.
Michael