ebroecker/canmatrix #403: XLSX export module: support for complex multiplexers

mwoco, 2019-10-08:
When converting DBC files to XLSX I get "ERROR - xlsx - Export complex multiplexers is not supported - ignoring frame …". Could you extent the XLSX export module with complex multiplex support? Thanks!

mwoco, 2019-12-06:
Thank you for the quick fix! I have been able to test the new version 0.9.1 and it's already working great as messages/signals are not dismissed anymore. ;)

I would like suggest one further modification: With the current code a can message like

 - Multiplexor A
 - Multiplexor B
 - Signal 1 (A = 1)
 - Signal 2 (A = 2)
 - Signal 3 (B = 1)
 - Signal 4 (B = 2)

turns into

 - Multiplexor A (Mode signal:)
 - Multiplexor B (Mode signal:)
 - Signal 1 (Mode 1:)
 - Signal 2 (Mode 2:)
 - Signal 3 (Mode 1:)
 - Signal 4 (Mode 2:)

within the xlsx-file. This way it is not possible to say anymore, which multiplexor (A or B) belongs to each signal. So I would recommend doing something like

 - Multiplexor A (Mode signal)
 - Multiplexor B (Mode signal)
 - Signal 1 (mode A = 1)
 - Signal 2 (mode A = 2)
 - Signal 3 (mode B = 1)
 - Signal 4 (mode B = 2)

Is that possible?

ebroecker, 2019-12-06:
@mwoco 
Thanks for testing.
Yes, this is why this issue is still open. Till now only the "abort" for "complex multiplexed" was removed.
I'll check your suggestion

ebroecker, 2019-12-14:
@mwoco did you try branch iss403

mwoco, 2019-12-26:
Sorry for the delay. I will give it a try at the beginning of January.

mwoco, 2020-01-07:
I've just tested branch iss403: But the output file looks the same like without your fix. Here is what I did:
 - download ZIP file from https://github.com/ebroecker/canmatrix/tree/iss403 an extract its contents to some directory
 - convert DBC by trying both "src\canmatrix\cli\convert.py" and "examples\convert.py"

ebroecker, 2020-01-07:
you need to install the new version first:
something like
`python setup.py install`

mwoco, 2020-01-07:
Ok. I used the install command and then tried to convert a DBC file using "canconvert.exe". This seems to work for a couple of messages and then fails with the following error message:

`Traceback (most recent call last):
  File "...\dbc2xlsx\WinPython64-3.7.4.1Zero\WPy64-3741\python-3.7.4.amd64\Scripts\canconvert-script.py", line 11, in <module>
    load_entry_point('canmatrix==0+unknown', 'console_scripts', 'canconvert')()
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\canmatrix-0+unknown-py3.7.egg\canmatrix\cli\convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\canmatrix-0+unknown-py3.7.egg\canmatrix\convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\canmatrix-0+unknown-py3.7.egg\canmatrix\formats\__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\canmatrix-0+unknown-py3.7.egg\canmatrix\formats\__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "...\dbc2xlsx\winpython64-3.7.4.1zero\wpy64-3741\python-3.7.4.amd64\lib\site-packages\canmatrix-0+unknown-py3.7.egg\canmatrix\formats\xlsx.py", line 260, in dump
    (frontRow, back_row) = canmatrix.formats.xls_common.get_signal(db, sig, motorola_bit_format)
TypeError: get_signal() missing 1 required positional argument: 'motorola_bit_format'`

mwoco, 2020-02-22:
Any update on this topic?

ebroecker, 2020-02-22:
oh - sorry - I forgot to have a look on it...
I'll need to find some time for having a look on it.

mwoco, 2020-04-24:
Had a chance to look on it meanwhile?

ebroecker, 2020-04-24:
no - sorry - not yet -

commit fe85c7b2 (2019-11-15): prepare release 0.9.1 (#435)

* add submodules doc
* update doc to make start_bit more clear in documentation #424
* [dbc] fix dbc issues (#431)
* [dbc] export for empty matrix
* copy_frame handles default values now (fix #430)
* [xlsx] export of coplex mutltiplexed frames no longer prohibited (fix #403)
closed 2020-09-29 by ebroecker