ebroecker/canmatrix #407: Signal Comments broke import

timtianyang, 2019-10-18:
Hi, I have a dbc that contains a block for signal comments. When I import the same dbc and export to dbc again, whatever contents after the comments are skipped.  This issue persists in merging as well. Can you take a look? Thanks!

An example comment line:
"CM_ SG_ 10 resistance "resistance setting (0-100%)" ;"

ebroecker, 2019-10-18:
Hi @timtianyang 

Thanks a lot for reporting this bug. I can already reproduce.
The problem is the space in front of the ";"

I am working on a fix

ebroecker, 2019-10-18:
could you try branch iss407

`pip install git+https://github.com/ebroecker/canmatrix.git@iss407`

timtianyang, 2019-10-18:
Thanks for the quick fix!
When I install it from git branch, it complains the following

Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 5, in <module>
    from canmatrix.cli.convert import main
ImportError: cannot import name main

ebroecker, 2019-10-19:
Maybe your installations collide. 

Could you remove your former installation, and try to call convert like this:
`python -m canmatrix.cli.convert` ([cli doc](https://canmatrix.readthedocs.io/en/latest/cli.html#general))

timtianyang, 2019-10-20:
Thanks. I can confirm all lines are picked up after signal comments. One more request, can it preserve the space before " ;" in all sections? I am trying to maintain the same contents before and after.

ebroecker, 2019-10-21:
There are lots of format variants which lead to the same canmatrix object.
Thus it is impossible to do some modifications with canmatrix and get the same format again.

But you can do a conversion from dbc to as first step, afterwards do the modification in a second step. 
Because canmatrix dbc exporter should export every time the same way, you can see the changes then.

```bash
canconvert source.dbc temp.dbc
canconvert --doSomeModification temp.dbc target.dbc
diff temp.dbc target.dbc
```

timtianyang, 2019-10-21:
thanks for the quick fix again! closing this out.

closed 2019-10-21 by timtianyang