ebroecker/canmatrix #410: Conversion arxml to dbc for CANFD messages

DeaPetrov, 2019-10-21:
I tried conversing an arxml to dbc file, for messages with addressing mode Extended and CAN FD Supported checked, it converts with ID format CAN extended. 
I also tried --setFrameFd command, but it works with maximum 2 messages at one. Can this limit also be changed to a higher limit, e.g.: 30 messages?
When doing the conversion, for FlexRay Frame, the messages are not exported to dbc file. Can also this type be converted?

DeaPetrov, 2019-10-22:
Hi, 
Thanks for the response.
I am beginner in Python, I tried to do the set up, so I can run the scripts. I still didn't manage to have it up and running.
Can you upload also convert.exe? 
I rechecked --setFrameFd command, it works with multiple messages if there is no space after ",".
Thank you!

ebroecker, 2019-10-22:
did you have a look on https://canmatrix.readthedocs.io/en/latest/installation.html?

ebroecker, 2019-11-07:
please reopen, if still an issue

closed 2019-11-07 by ebroecker