ebroecker/canmatrix #411: Multiplexer is appended to signal name

timtianyang, 2019-10-21:
It looks like the multiplexer number is automatically appended at the end of the signal name after a conversion. For example,

```
BO_ 1284 ECUState: 8 ECU
 SG_ ECU_Mul M : 0|2@1+ (1,0) [0|3] "enum" DBG
 SG_ ECU_class m0 : 2|2@1+ (1,0) [0|3] "enum" DBG
```

Will get converted into

```
BO_ 1284 ECUState: 8 ECU
 SG_ ECU_Mul M : 0|2@1+ (1,0) [0|3] "enum" DBG
 SG_ ECU_class0 m0 : 2|2@1+ (1,0) [0|3] "enum" DBG
```
This can cause issues with automatically generated code as applications expect the same way to access the signal by using the name without the multiplexer. Is there anyway to turn this feature off?

ebroecker, 2019-10-22:
Hi @timtianyang,

this only happens, if you have the same signal name more than once in one frame.
If the signal name is not unic, a postfix index is appended.
This does not happen depending on the multiplexer.
```
BO_ 1284 ECUState: 8 ECU
 SG_ ECU_Mul M : 0|2@1+ (1,0) [0|3] "enum" DBG
 SG_ ECU_class m0 : 2|2@1+ (1,0) [0|3] "enum" DBG
```
keeps untouched with canmatrix.cli.convert,

but
```
BO_ 1284 ECUState: 8 ECU
 SG_ ECU_Mul M : 0|2@1+ (1,0) [0|3] "enum" DBG
 SG_ ECU_class m0 : 2|2@1+ (1,0) [0|3] "enum" DBG
 SG_ ECU_class m1 : 10|2@1+ (1,0) [0|3] "enum" DBG
```
will get to 
```
BO_ 1284 ECUState: 8 ECU
 SG_ ECU_Mul M : 0|2@1+ (1,0) [0|3] "enum" DBG
 SG_ ECU_class0 m0 : 2|2@1+ (1,0) [0|3] "enum" DBG
 SG_ ECU_class1 m1 : 10|2@1+ (1,0) [0|3] "enum" DBG
```

If you have a real live use case, where the same signal name is used multiple in one frame, we could add an option to disable the *signal unification*. 

Note:
Usually the toolchains which use the dbc are not happy with this. Also signal names are used to identifiy additional information like comment or attributes to the signal. This will lead to strange behaviour afterwards.

timtianyang, 2019-10-22:
Hi @ebroecker ,
I do have a real live use case, where people might want to see the same signal with different multiplexers, without carrying around the multiplexer after the signal name. This makes auto generated code more maintainable in case the multiplexers are changed.

ebroecker, 2019-10-23:
@timtianyang 

please try branch 411:
for example:
`pip install git+https://github.com/ebroecker/canmatrix.git@iss411`

there you'd have a new command line option:
`--no-dbcUniqueSignalNames`

```bash
python -M canmatrix.cli.convert --no-dbcUniqueSignalNames source.dbc target.dbc
```

closed 2019-10-25 by timtianyang

timtianyang, 2019-10-25:
are you planning to merge this back to master?