ebroecker/canmatrix #421: extracting ecus from dbc based on sender/receiver

timtianyang, 2019-10-30:
Is it possible to extract ecus based on receiver and sender separately?  Like "--ecu_sender=MyEcu1 --ecu_receiver=MyEcu2" will extract all frames sent by MyEcu1 and all frames received by MyEcu2 into a new dbc.

btw, this option is in the code: https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/convert.py#L108
but looks like it's missing cli document.

ebroecker, 2019-10-30:
addFrameReceiver is here:
https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/cli/convert.py#L65

there is no option for ecu_sender/receiver selection yet. Maybe I find some time...

ebroecker, 2019-10-31:
branch iss421;

syntax for your example would be:
`--ecus=MyEcu1:tx,MyEcu2:rx`

timtianyang, 2019-11-01:
Thanks for such a quick turn around time!

ebroecker, 2019-11-01:
so - it works as expected?

ebroecker, 2019-11-03:
merged to development

closed 2019-11-03 by ebroecker

timtianyang, 2019-11-04:
yes it's working as expected from my tests.