ebroecker/canmatrix #424: Motorola signals bug

MatinF, 2019-11-06:
Hi canmatrix team,

We seem to encounter an issue when trying to load DBC files with Motorola byte order, e.g. as below signal example:

```
BO_ 1709 CAN_ID_NAME: 8 Control
 SG_ SIGNAL_NAME: 6|1@0+ (1,0) [0|1] ""  XYZ
```
If I manually hardcode the above to replace `@0+` with `@1+` it provides the correct result for this particular signal, but presumably it would be wrong for cross-byte signals in that case.

Let me know if you need further details.

danielhrisca, 2019-11-06:
This is how the message looks like in Vector's CANdb++ Editor

[image]

and the actual dbc entry:

```
BO_ 1809 MSG: 8 DEV2
 SG_ SIG1 : 39|4@0+ (1,0) [0|1] ""  DEV1
 SG_ SIG2 : 52|1@0+ (1,0) [0|1] ""  DEV1
 SG_ SIG3 : 51|12@0+ (0.1,0) [0|360] "°"  DEV1
 SG_ SIG4 : 6|1@0+ (1,0) [0|1] ""  DEV1
 SG_ SIG5 : 5|1@0+ (1,0) [0|1] ""  DEV1
 SG_ SIG6 : 23|3@0+ (1,0) [0|1] ""  DEV1
 SG_ SIG7 : 7|1@0+ (1,0) [0|1] ""  DEV1
 SG_ SIG8 : 34|11@0+ (0.1,-102.4) [-32|32] "A"  DEV1
 SG_ SIG9 : 18|11@0+ (0.1,-102.4) [-62.5|62.5] "A"  DEV1
 SG_ SIG10 : 4|13@0+ (0.1,0) [350|450] "V"  DEV1
```

ebroecker, 2019-11-06:
Wow - I hoped all these little/big endian issues are behind us...

Thanks for your report.

I'll need some time to get into byte order and dbc it again to understand whats wrong here.

danielhrisca, 2019-11-06:
@ebroecker 
the patch from my PR produces the same start bit values as Vector CANdb++; I don't know if this fits with the canmatrix internal logic

ebroecker, 2019-11-06:
OK, there seems to be a bug in `get_starbit` for `bit_numbering=True` and `start_little=True`.
Your patch exactly fixes this conditions.

oh - I found a comment in the tests... `# TODO add test for reversed endianning` :-)

Thanks.

[image]
[image]

ebroecker, 2019-11-06:
Well - now I am completely confused.

After creating some tests, it seems to work correct at my side.

I'll add some tests and push, than you can compare.

ebroecker, 2019-11-06:
@MatinF @danielhrisca 

could you please have a look on https://github.com/ebroecker/canmatrix/commit/afc66767b6b846aa8957268e6f98e2715cea63d3, especially to `test_candbpp_startbit`.

These tests seam to run without error. 
Where do we do an mistake?

Thanks,
Eduard

danielhrisca, 2019-11-06:
@ebroecker 
the expected values used in test_candbpp_startbit do not match with the values from the Vector CANdb++ screenshots

ebroecker, 2019-11-06:
```python
    assert matrix.frames[0].signal_by_name("SIG1").get_startbit(True, True) == 36
    assert matrix.frames[0].signal_by_name("SIG2").get_startbit(True, True) == 52
    assert matrix.frames[0].signal_by_name("SIG3").get_startbit(True, True) == 56
    assert matrix.frames[0].signal_by_name("SIG4").get_startbit(True, True) == 6
    assert matrix.frames[0].signal_by_name("SIG5").get_startbit(True, True) == 5
    assert matrix.frames[0].signal_by_name("SIG6").get_startbit(True, True) == 21
    assert matrix.frames[0].signal_by_name("SIG7").get_startbit(True, True) == 7
    assert matrix.frames[0].signal_by_name("SIG8").get_startbit(True, True) == 40
    assert matrix.frames[0].signal_by_name("SIG9").get_startbit(True, True) == 24
    assert matrix.frames[0].signal_by_name("SIG10").get_startbit(True, True) == 8
```
[image]

where exactly - seems I am blind

danielhrisca, 2019-11-06:
I was looking at the Intel asserts. Sorry

ebroecker, 2019-11-06:
@MatinF @danielhrisca

I can't find a bug here.
```python
.get_startbit(True, True)
```
gives you the start_bit as you want it - at least with the examples you provided.

could you please verify ior clarify ?

Thanks

danielhrisca, 2019-11-06:
Hello Eduard,

I've installed the latest canmatrix development code and added this snippet

```python
                    print(frame.name, temp_signal.name, temp_signal.start_bit)
```

after this line (after the call to set_startbit) https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/formats/dbc.py#L574

and I get this print 

```
MSG SIG1 32
MSG SIG2 51
MSG SIG3 52
MSG SIG4 1
MSG SIG5 2
MSG SIG6 16
MSG SIG7 0
MSG SIG8 37
MSG SIG9 21
MSG SIG10 3

```

ebroecker, 2019-11-06:
what about:
```python
print(frame.name, temp_signal.name, temp_signal.get_startbit(True, True))
```

danielhrisca, 2019-11-06:
```
Start bit     MSG SIG1 32
Get start bit MSG SIG1 36
Start bit     MSG SIG2 51
Get start bit MSG SIG2 52
Start bit     MSG SIG3 52
Get start bit MSG SIG3 56
Start bit     MSG SIG4 1
Get start bit MSG SIG4 6
Start bit     MSG SIG5 2
Get start bit MSG SIG5 5
Start bit     MSG SIG6 16
Get start bit MSG SIG6 21
Start bit     MSG SIG7 0
Get start bit MSG SIG7 7
Start bit     MSG SIG8 37
Get start bit MSG SIG8 40
Start bit     MSG SIG9 21
Get start bit MSG SIG9 24
Start bit     MSG SIG10 3
Get start bit MSG SIG10 8```

ebroecker, 2019-11-06:
so - get_startbit gives you the correct value?

danielhrisca, 2019-11-06:
> so - get_startbit gives you the correct value?

Yes, it looks correct

ebroecker, 2019-11-08:
so we can close this bug?

danielhrisca, 2019-11-08:
The fact remains that using the load dbc function produces wrong output

ebroecker, 2019-11-08:
I don't understand at the moment.

YES, canmatrix internally has another startbit logic.

BUT there is the `get_startbit` method which calculates the start_bit in other formats.
If you use it, you get the right values. This at least was the conclusion of the posts above?

So if you need some other start_bit logic you have to use `get_startbit`.

By the way, start_bit representation is configurable in candb++ also.

Do I miss something?

```python
#!/usr/bin/env python3

import canmatrix.formats

db = canmatrix.formats.loadp_flat("bb.dbc")

for signal in db.frames[0].signals:
    print( signal.name, signal.get_startbit(True,True))
```
[bb.dbc.txt](https://github.com/ebroecker/canmatrix/files/3824818/bb.dbc.txt)

[image]

danielhrisca, 2019-11-08:
I was not aware that the start_bit is internal. Maybe it should be implemented as a property

ebroecker, 2019-11-08:
There are different needs for the `start_bit` depending on the usecase. 
Thus once @dmahurin fixed the behaviour to have one representation in `start_bit` which is use quite common. For all other needed representations he created the `get_startbit` which helps out.

altendky, 2019-11-08:
> I was not aware that the start_bit is internal. Maybe it should be implemented as a property

What about being internal indicates it should be a `@property`?

danielhrisca, 2019-11-08:
In Python land the coomon practice is to prepend an underscore to internal variables.

With the property approach I was thinking about hiding the start bit internal processing

ebroecker, 2019-11-08:
We had a lot confusion already regarding start_bits.
`internal` is the wrong word anyway (my fault), we think, our `start_bit` is quite *consistent* whereas dbc is not consistent:

...
With "consistent" format/numbering, bit numbering follows the byte format. MSB0 for big endian, LSB0 for little endian. Please refer our [wiki_page](https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder).

ebroecker, 2020-02-04:
please reopen if you still think this is a bug

closed 2020-02-04 by ebroecker