ebroecker/canmatrix #428: extraction based on attributes

timtianyang, 2019-11-07:
Is there anyway to filter/extract frames based on attributes?
python -m canmatrix.cli.convert --attr=my_attribute_name:attr_value source.dbc filtered.dbc

timtianyang, 2019-11-08:
looks like a simple function addition for:
`def frame_by_attribute(self, attr)`
and then 
`def copy_frame(attr, source_db, target_db):`
would do. Not sure how you like to handle the code duplication though.

ebroecker, 2019-11-08:
maybe meanwhile you can do some script yourself:
```python
import canmatrix.formats
import canmatrix.copy

target = canmatrix.CanMatrix()
cm = canmatrix.formats.loadp_flat("some.dbc")
copy_frames = [f for f in cm.frames if f.attributes.get("YOURATTRIBUTE", "") == "YOURVALUE"]

for frame in copy_frames:
  canmatrix.copy.copy_frame(cm, target)

canmatrix.formats.dump({"":target},"out.dbc")
```

Note Tested !

timtianyang, 2019-11-08:
thanks for the suggestion.

It looks like attribute parsing has some issue with a white space in front of ';' add attributes.get() will not find it.

Also, if I remove the white space, the string attribute got appended with an extra " at the end.

test case:

The following will fail the parsing
BA_ "Bus" BO_ 512 "BDY2" ;                                                                                                            
BA_ "Bus" BO_ 513 "BDY2" ;

The following
BA_ "Bus" BO_ 512 "BDY2";                                                                                                            
BA_ "Bus" BO_ 513 "BDY2";
 turns into
BA_ "Bus" BO_ 512 "BDY2"";
BA_ "Bus" BO_ 513 "BDY2"";

ebroecker, 2019-11-08:
thanks for this issue, gonna have a look ...

timtianyang, 2019-11-08:
I confirm that the white space issue has been fixed by 961918a6 on branch iss427.

ebroecker, 2019-11-09:
my code was not handling default values, so here's an update:

```python
import canmatrix.formats
import canmatrix.copy

target = canmatrix.CanMatrix()
cm = canmatrix.formats.loadp_flat("some.dbc")
copy_frames = [f for f in cm.frames if f.attribute("YOURATTRIBUTE", cm) == "YOURVALUE"]

for frame in copy_frames:
  canmatrix.copy.copy_frame(cm, target)

canmatrix.formats.dump({"":target},"out.dbc")
```

use frame.attribute("YOURATTRIBUTE", matrix) to get the atrribute value or the default.
https://canmatrix.readthedocs.io/en/latest/api.html#canmatrix.canmatrix.Frame.attribute

closed 2021-05-07 by ebroecker