ebroecker/canmatrix #430: missing attribute defaults

timtianyang, 2019-11-08:
test case:

The following attribute and default values will not get picked up unless some frames have specified values for the attribute.

For example, the following will get skipped:
```
BA_DEF_ BO_ "GenMsgDelayTime" INT 0 65535;
BA_DEF_DEF_ "GenMsgDelayTime" 0 ;
```
unless a message has non default value assigned:
`BA_ "GenMsgDelayTime" BO_ 513 100 ;`

ebroecker, 2019-11-08:
please try branch `iss427`

timtianyang, 2019-11-08:
I think iss427 fixes the white space issue, but defaults are still missing.

ebroecker, 2019-11-09:
well - it is fixed also, but handling of default values is different...
I'll update the example-code in issue #428 to show how to handle with 
default values, ...

timtianyang, 2019-11-11:
Thanks for the example. I think issue #430 is still persists. It's not about filtering out default values. It's about copying frames, but some default values are not copied, unless specified with a non default value.

ebroecker, 2019-11-12:
should be fixed now. (in branch `iss427`)

timtianyang, 2019-11-12:
thanks for the fix!

ebroecker, 2019-11-12:
@timtianyang 
Thanks for reporting all these bugs

ebroecker, 2019-11-19:
sometimes it happens that "None" is exported as attribute.

commit b0b36bf5 (2019-11-12): copy_frame handles default values now (fix #430)
commit f1619656 (2019-11-12): [dbc] fix dbc issues (#431)

* fix dbc export for empty matrix
* copy_frame handles default values now (fix #430)
commit fe85c7b2 (2019-11-15): prepare release 0.9.1 (#435)

* add submodules doc
* update doc to make start_bit more clear in documentation #424
* [dbc] fix dbc issues (#431)
* [dbc] export for empty matrix
* copy_frame handles default values now (fix #430)
* [xlsx] export of coplex mutltiplexed frames no longer prohibited (fix #403)
closed 2019-11-21 by ebroecker