ebroecker/canmatrix #433: any plan to export the signal group attribute out from the arxml?

sky-dream, 2019-11-13:
any plan to export the signal group attribute out from the arxml when export the CAN signal info or Flexray info from the arxml file?

ebroecker, 2019-11-13:
well - it's possible - do you have an usecase for it? If so, we can do.

sky-dream, 2019-11-15:
hi , ebroecker, tks very much for your help, since out ECU in the network also need to play as gateway to forward the signal groups between the Backbone network(Flexray) and the domain network(CAN network), so we hope the signal group , PDU, PDU group attributes can also be exported to let us check the network status more easily.

sky-dream, 2019-11-15:
hi , ebroecker, tks very much for your help, since our ECU in the network also need to play the  gateway role to forward the signal groups between the Backbone network(Flexray) and the domain network(3 CAN networks), so we hope the signal group , PDU, PDU group attributes can also be exported to let us check the network status more easily.

ebroecker, 2019-11-15:
we'll do this further in #432

closed 2019-11-15 by ebroecker