ebroecker/canmatrix #434: BU_ not updated when deleting a frame

timtianyang, 2019-11-14:
canconvert --deleteFrame=my_frame input.dbc output.dbc

If the message to be deleted has either rx or tx associated with a node, and it is the only instance of appearing for that node in a dbc, it should also be removed from BU_ list, as well as CM_ BU_ section.

For example:

BU_ : ECU1 ECU2 ECU3
...
BO_ xxx my_frame: 8 ECU1
  xxx

...
CM_ BU_ ECU1: "My ecu1";

where ECU1 is only appearing as tx node for my_frame in that dbc. It should be removed in both places(BU_, and CM_ BU_)

ebroecker, 2019-11-14:
There may be use cases where someone wants to add a new frame again, which uses this ecu.
So at least sometimes automatic deleting would not be ok.

Meanwhile you could use `--deleteEcu` and do it manually.

Maybe I'll add some option like `--deleteUnusedEcus` - if this would fulfill your needs

timtianyang, 2019-11-15:
Gotcha. You are absolutely right. 

Side question, what's the release cadence? I wanted to use canmatrix with my projects but it looks like some last minute bug fixes weren't in the last release. I'm using bazel and it would require some efforts to write build rules for this project and put that on CI, pointing to the development branch.

ebroecker, 2019-11-15:
gonna create a release this weekend, not yet have enough practice to do high frequently...

ebroecker, 2019-11-15:
release 0.9.1 is out at pypi

@timtianyang 
do you have a need for `--deleteUnusedEcus`? If not, I'll close this issue

ebroecker, 2019-11-17:
`--deleteObsoleteEcus` is in branch `iss434`

ebroecker, 2019-11-21:
merged to development

closed 2019-11-21 by ebroecker