ebroecker/canmatrix #436: Loading arxml fails in 0.9.1

uruuru, 2019-11-20:
Trying to load an arxml fails in the latest release (it worked in 0.8.0). 
The following line yields an error that `speed` is a bool and thus has no `.text` property:
https://github.com/ebroecker/canmatrix/blob/d537f9c6d381d036eae7dde4c21f58ba0e35d1ce/src/canmatrix/formats/arxml.py#L1701

`speed` is initialized a couple of lines before:
https://github.com/ebroecker/canmatrix/blob/d537f9c6d381d036eae7dde4c21f58ba0e35d1ce/src/canmatrix/formats/arxml.py#L1691

The issue seems to have been introduced here: 
https://github.com/ebroecker/canmatrix/commit/9aa5db7c25b00cb5cfd244361f5eb0e440601f41#diff-caaa1c3759a631dfbc62376ad283e206R1691

ebroecker, 2019-11-20:
@uruuru 
Thanks for reporting this issue, gonna provide a fix soon.

ebroecker, 2019-11-20:
please try branch `iss436`

uruuru, 2019-11-21:
Thanks for the fast fix 👍 For mr it seems to work.

ebroecker, 2019-11-21:
Thanks for testing

closed 2019-11-21 by ebroecker