ebroecker/canmatrix #443: Usecases: Please Add here

ebroecker, 2019-12-06:
Hi everybody,
you are welcome to add your use case of canmatrix here.
I just want to have some idea what people are using canmatrix for.

So feel free to tell something you use it for or "vote" some comment.

Thanks,
I'll take the answers here into account for further prioritization of issues.

ForestRupicolous, 2020-01-07:
Hello, I would like to use canconvert to convert an .arxml AUTOSAR V 4.2.2 to a dbc (in fact multiple as there are multiple network defined).  Is support for AUTOSAR version other than only 3.2.3 and                       4.1.0 planned?

ebroecker, 2020-01-10:
Hi @ForestRupicolous 
Maybe you did not install the required packages.
How did you install? Did you read https://canmatrix.readthedocs.io/en/latest/installation.html?

ForestRupicolous, 2020-01-10:
Hi @ebroecker,
thank you for your fast response. You are correct I totally missed these packages. I installed via pip and was expecting that all required modules where added as some where automatically installed.
Installed them now. I have to check if the conversion is working next week as I haven't the file available on this PC.

krishomprasad, 2020-02-01:
Hi , @ebroecker , 

Currently the  cancompare outputs the difference in the console, can it output in excel with value of old and new , it would helpful to know what exactly the current output is helpful it would more helpful if we were to view the output in spread sheet with the differences atleast for the signals(new and old value) / messages( whether periodicity or receiver/transmitter attributes got modified) .

Seneda, 2020-04-07:
Hi @ebroecker 

Thanks for writing this library. I am using canmatrix for handling Symbol files which my company generates for our products and for converting to other formats which our users need.
I am also finding it useful for manipulating Symbol files to add duplicate frames (with incrementing IDs etc.) and things where being able to manipulate things in python is far easier than editing them either by hand or in the Symbol Editor from PEAK. So having .sym support is very useful.

I have noticed that there are some features missing from the Symbol format parsing/writing. (particularly around signal types and enumerations). I've been able to modify the code to do what i need, are you looking for more contributions?

Thanks
Seneda

ebroecker, 2020-04-08:
@Seneda 

You are very welcome to add your contributions here.

Just create a pullrequest...

Seneda, 2020-04-08:
Another use case i have is to use the .json format for interfacing with tools I write in other languages (as json parsing libraries already exist and it is much simpler to not have to write a .sym parser in C# for example).
There are some things missing the .json files generated compared with the .sym file, enumeration value tables for example. I can easily make changes to add these, but is the .json structure used here based on a standard CAN db format or is it unique to this library? I wouldn't want to try and make any changes that would break a dependency of some other library or standard etc.

ebroecker, 2020-04-08:
@Seneda 

You are free to do so. If you look at #136 (very old issue) - I wanted to add data to json.

Just add these things to the "export all" section of the json formater. 
Maybe - importing these new items would be great also :-)

yangx1990, 2021-03-02:
hello @ebroecker 

I noticed that the comment in arxml.py says "currently support for AUTOSAR 3.2 and 4.0-4.3 is planned". 
Can I use this tool to convert arxml(AUOTSAR 4.2) file to DBC file now?
When I converted the arxml file, I found that the transmitter of the frame was not generated into the DBC file.
May I ask you for some advice？

ebroecker, 2021-03-02:
Hi @yangx1990 ,

yes you can. should work out of the box.
please let me know if problems occure

yangx1990, 2021-03-03:
@ebroecker 
Thank you for your response.

There are 2 arxml files in my hand.
In the element "ASSOCIATED-COM-I-PDU-GROUP-REFS" of one arxml file, customer used "CONTAINED-I-SIGNAL-I-PDU-GROUP-REFS" while I found "CONTAINED-I-PDU-GROUPS-REFS" is used in arxml.py. And then "I-PDU-REFS" in arxml.py is instead of "I-SIGNAL-I-PDUS" in my arxml file.
In the other arxml file, no "ASSOCIATED-COM-I-PDU-GROUP-REFS" or "ASSOCIATED-I-PDU-GROUP-REFS" could be found.
I think that may be the reason why frame transmitter can not be generated when I used canmatrix to convert arxml file to dbc file.

Joshua-Calvert, 2021-04-29:
In our project, we have to maintain CAN databases in IBM DOORS. This tool is able to import/export with Microsoft Excel. To generate DBC-files I thought of canmatrix's DBC<->Excel conversion features as an intermediate tool from DOORS over EXCEL to DBC.

To evaluate the conversion, I tried to "cancovert.exe" an ordinary DBC file to XLS and back to DBC. The XLS looks fantastic and I was very happy with the result. But the back conversion failed, min/max and scaling information for the signals are ignored. I think, this is not a bug, it is just a missing feature according to the [Supported File Fomats list](https://canmatrix.readthedocs.io/en/latest/formats.html). What does the "p" in the table stand for? Planned? I would really appreciate that.

ebroecker, 2021-04-30:
hI @Joshua-Calvert 
xls(x) is a "lossy" format.
"p" means "partly".

What you need is a xlsx-format which does not loose information.
It can be done.
Which attributes you need?

maybe you could try during export and import:
--additionalSignalAttributes is_signed,factor,offset

Morgan-Swanson, 2021-05-21:
I'm looking to convert the SAE digital annex excel file into a dbc format. Unfortunately I can't find a tool to do this.

AhmedKotb10, 2023-04-17:
Hi @ebroecker 
is it available to use the library to get ethernet cluster information?

ebroecker, 2023-07-04:
Hi @kruti2107 

yes, a little bit.
Try this command line option:
`--arxmlEthernet`

kruti2107, 2023-07-04:
@ebroecker. Thanks for the quick reply.

Is this the only way with command line? Is there any function or something like that which could I include in my code?

ebroecker, 2023-07-04:
Hi @kruti2107 ,

for sure, you can.
I don't know your code, but if you do something like:

`cluster = canmatrix.formats.arxml.load(arxml, decode_ethernet=True)`

the information which is extracted is very limited:
https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/formats/arxml.py#L1745

kruti2107, 2023-07-06:
Hi @ebroecker ,

After the arxml load oepration, I am getting empty object. No information related to my ethernet ARXML.

Do you have any idea, what could be the possible reason?

kruti2107, 2023-07-06:
Hi @ebroecker , It was a fault from my side. I made some wrong configuration. It's done now. 

Thanks for the suggestion

ebroecker, 2023-07-06:
Hi @kruti2107 

perfect.
Does this little extracted information help you?

kruti2107, 2023-07-07:
@ebroecker  Not sure right now, how much will it help. I was looking for way to extract info from the db.

wxie1, 2023-08-25:
> In our project, we have to maintain CAN databases in IBM DOORS. This tool is able to import/export with Microsoft Excel. To generate DBC-files I thought of canmatrix's DBC<->Excel conversion [...]

I have done the similar:
 "cancovert.exe" an ordinary DBC file to XLS and back to DBC, but

"cancovert.exe" an ordinary DBC file to XLS works without problem.
xls(x) back to DBC does not work, and it says: format not supported.

May I ask how you did the back to DBC part?