ebroecker/canmatrix #449: Extracting signal from DBC without referencing frame name

MatinF, 2020-02-14:
Hi there,

I'm looking to extract signal properties from a DBC by parsing the signal name - without having to parse other information.

Currently I'm able to achieve the extraction via below:

```
dbc_path = Path(__file__).parent / "j1939.dbc"
message_name = "EEC1"
signal_name = "EngineSpeed"
dbc = canmatrix.formats.loadp_flat(dbc_path, "dbc")
message = dbc.frame_by_name(message_name)
signal = message.signal_by_name(signal_name)
```

However, what I would ideally want would look similar to below:
```
dbc_path = Path(__file__).parent / "j1939.dbc"
signal_name = "EngineSpeed"
dbc = canmatrix.formats.loadp_flat(dbc_path, "dbc")
signal = dbc.signal_by_name(signal_name)
```

Is this possible via some function/API method? I've tried reviewing the docs, but did not find anything. I'm sure some manual search process could be set up, but just wanted to check if there was a built-in method for this.

Martin

ebroecker, 2020-02-14:
Hi @MatinF ,

there is no such option (yet).

you could do it this way, not nice anyway:
```python
[f.signal_by_name("EngineSpeed") for f in dbs["CAN"].frames if f.signal_by_name("EngineSpeed") is not None]
```

This gives you an array with probably one entry.

MatinF, 2020-02-14:
Great, it worked by using below for me:

```
[f.signal_by_name("EngineSpeed") for f in dbs.frames if f.signal_by_name("EngineSpeed") is not None]
```

closed 2020-02-14 by MatinF