ebroecker/canmatrix #450: Question: How to formulate extended id acceptable?

anterosa, 2020-02-18:
My Script is copy paste from decode example.

Working with this set up:

_pi@raspberrypi:~ $ pip3 show canmatrix
Name: canmatrix
Version: 0.9.1
Summary: UNKNOWN
Home-page: http://github.com/ebroecker/canmatrix
Author: None
Author-email: None
License: BSD
Location: /home/pi/.local/lib/python3.5/site-packages
Requires: bitstruct, click, future, attrs, pathlib2
pi@raspberrypi:~ $ python3 -V
Python 3.5.3_

I get following error while trying to run script:

_pi@raspberrypi:~ $ python3 canmatrix_test.py STAGEV_DPF.dbf 62526#5555555555555555
Traceback (most recent call last):
  File "canmatrix_test.py", line 21, in <module>
    db = canmatrix.formats.loadp_flat(sys.argv[1])
  File "/home/pi/.local/lib/python3.5/site-packages/canmatrix/formats/__init__.py", line 85, in loadp_flat
    dbs = loadp(path, import_type, key, **options)
  File "/home/pi/.local/lib/python3.5/site-packages/canmatrix/formats/__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "/home/pi/.local/lib/python3.5/site-packages/canmatrix/formats/__init__.py", line 96, in load
    dbs[key] = module_instance.load(file_object, **options)  # type: ignore
  File "/home/pi/.local/lib/python3.5/site-packages/canmatrix/formats/dbf.py", line 222, in load
    new_frame.arbitration_id = canmatrix.ArbitrationId.from_compound_integer(int(arb_id))
  File "/home/pi/.local/lib/python3.5/site-packages/canmatrix/canmatrix.py", line 699, in from_compound_integer
    extended=(i & cls.compound_extended_mask) != 0,
  File "<attrs generated init canmatrix.canmatrix.ArbitrationId>", line 4, in __init__
  File "/home/pi/.local/lib/python3.5/site-packages/canmatrix/canmatrix.py", line 599, in __attrs_post_init__
    raise ArbitrationIdOutOfRange('ID out of range')
canmatrix.canmatrix.ArbitrationIdOutOfRange: ID out of range_

Snippet from dbf file:

//******************************BUSMASTER Messages and signals Database ******************************// 

[DATABASE_VERSION] 1.3 

[PROTOCOL] J1939 

[BUSMASTER_VERSION] [3.2.2] 
[NUMBER_OF_MESSAGES] 28 

[START_MSG] DM1,62526,8,6,1,X, 
[START_SIGNALS] FMI,5,5,0,U,31,0,1,0,1,,, 
[START_SIGNALS] SPN,19,5,5,U,524287,0,0,0,1,,, 
[START_SIGNALS] SVS,2,1,0,I,1,-2,1,0,1,,, 
[START_SIGNALS] RS,2,1,4,U,3,0,1,0,1,,, 
[START_SIGNALS] AW,2,1,2,U,3,0,1,0,1,,, 
[START_SIGNALS] Mil,2,1,6,U,3,0,1,0,1,,, 
[END_MSG] 

[START_MSG] DM32,41472,8,3,1,X, 
[START_SIGNALS] FMI,5,3,0,U,31,0,1,0,1,-,, 
[START_SIGNALS] SPN,16,1,0,U,65535,0,1,0,1,-,, 
[VALUE_DESCRIPTION] "5839 - Interruption of dosing,5839",0 
[START_SIGNALS] DFCx_total_active_time,16,4,0,U,65535,0,1,0,0.2,h,, 
[END_MSG]

ebroecker, 2020-02-18:
Hi @anterosa ,

Thanks for your report.
Looks like you found a bug in dbf.

just to be sure, did you create the dbf with Busmaster?

ebroecker, 2020-02-18:
you created a arbitration_id `0xF43E` which is not allowed at can (max is `0x7FF`).
This is because canmatrix did not reccognize that you created a j1939 dbf.

I'll add some support...

ebroecker, 2020-02-18:
please try branch 450:

https://github.com/ebroecker/canmatrix/tree/iss450

`pip install git+https://github.com/ebroecker/canmatrix.git@iss450`

anterosa, 2020-02-21:
Hello,

Sorry for slow reply. Yes I created dbf with Busmaster.
Thanks for reply!

ebroecker, 2020-03-03:
@anterosa 

did you try the branch?

closed 2020-09-20 by ebroecker