ebroecker/canmatrix #453: ARXML: Signal name wrong

ForestRupicolous, 2020-02-25:
When I'm converting an .arxml to a .dbc the signal names are wrongly converted. I'm not sure about the correct AUTOSAR term but I think the system-signal-name is used instead of the I-signal name. In my case they differ by a leading "I", e.g. "ImySignal", vs. "mySignal". In the test file for arxml.py (src\canmatrix\tests\ARXMLSecuredPDUTest.arxml) they are the same and therefore the test passes.

I think I already tracked down the issue in the code (in arxml.py):
```python
        name = get_child(system_signal, "SHORT-NAME", root_or_cache, ns)  
```
should be:
```python
        name = get_child(isignal, "SHORT-NAME", root_or_cache, ns)
```

The next lines for context:
```python
        name = get_child(system_signal, "SHORT-NAME", root_or_cache, ns)
        unit_element = get_child(isignal, "UNIT", root_or_cache, ns)
        display_name = get_child(unit_element, "DISPLAY-NAME", root_or_cache, ns)
```

Am I correct with my assumptions? Are you interested in a pull-request?

ForestRupicolous, 2020-02-25:
Ok, I fear my understanding of the code doesn't go far enough for a pull-request. I just tried to update the tests to show the issue. But then I get a lot of other test errors I'm currently not able to fix.

ebroecker, 2020-02-26:
@ForestRupicolous 

Thanks for your input, I'll have a look on it the next days.

ebroecker, 2020-04-26:
You want the i-signal-shortnames instead of the system-signal-shortnames for the result singal name, right?
I think this depends on your wishes.
Currently I use the system-signal-shortnames.

But I currently plan a big rework of the arxml decoder.
Maybe I could do this configurable.

ebroecker, 2021-05-07:
fixed now with #571

it is configurable in convert now:
`--signalNameFromAttrib [SysSignalName|ISignalName|CompuMethodName]`

ebroecker, 2022-03-31:
should be fixed, otherwise please shout!

closed 2022-03-31 by ebroecker