ebroecker/canmatrix #454: Decimal exception while converting arxml to dbc

jungwoo-jeff-heo, 2020-02-26:
Hi, I am using canmatrix for converting arxml to dbc.

I have an issue as the following Traceback

`Traceback (most recent call last):
  File "examples\convert.py", line 6, in <module>
    canmatrix.convert.main()
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\convert.py", line 402, in main
    convert(infile, outfileName, **cmdlineOptions.__dict__)
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\convert.py", line 43, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\formats.py", line 64, in loadp
    return load(fileObject, importType, key, flatImport, **options)
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\formats.py", line 74, in load
    dbs = moduleInstance.load(fileObject, **options)
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1612, in load
    frameObject = getFrame(frameTrig,searchPoint,multiplexTranslation,ns, float_factory)
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1369, in getFrame
    getSignals(pdusigmapping, newFrame, xmlRoot, ns, None, float_factory)
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 1118, in getSignals
    (values, factor, offset, unit, const) = decodeCompuMethod(compmethod, xmlRoot, ns, float_factory)
  File "C:\Python27\lib\site-packages\canmatrix-0+unknown-py2.7.egg\canmatrix\arxml.py", line 983, in decodeCompuMethod
    if ll is not None and desc is not None and int(float_factory(ul.text)) == int(float_factory(ll.text)):
  File "C:\Python27\lib\decimal.py", line 547, in __new__
    "Invalid literal for Decimal: %r" % value)
  File "C:\Python27\lib\decimal.py", line 3872, in _raise_error
    raise error(explanation)
decimal.InvalidOperation: Invalid literal for Decimal: '0b0000000000000000'`

I used canmatrix for a while without issue, but I faced this issue after replacing the PC.
Please let me know how I can solve it

Best Regards
Jungwoo Heo

ebroecker, 2020-02-26:
@jungwoo-jeff-heo 
Thanks for your report.

give me some days to dig into thid issue.

ebroecker, 2020-03-01:
could you please try branch `arxml-limits-number-formats`

`pip install git+https://github.com/ebroecker/canmatrix.git@arxml-limits-number-formats`

jungwoo-jeff-heo, 2020-03-02:
Dear @ebroecker 

Thanks for updating, however it still has a problem with the following traceback.

1> Python 2.7
`C:\Git\canmatrix>python examples\convert.py examples\7.1_OMS_STAR_3_2020_d_Ecu_Details_Extract_2019_05a0_AR43.arxml 7_1.dbc
INFO - convert - Importing examples\7.1_OMS_STAR_3_2020_d_Ecu_Details_Extract_2019_05a0_AR43.arxml ...
Traceback (most recent call last):
  File "examples\convert.py", line 6, in <module>
    canmatrix.cli.convert.cli_convert()
  File "C:\Python27\lib\site-packages\click-7.0-py2.7.egg\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Python27\lib\site-packages\click-7.0-py2.7.egg\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Python27\lib\site-packages\click-7.0-py2.7.egg\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Python27\lib\site-packages\click-7.0-py2.7.egg\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Python27\lib\site-packages\canmatrix\cli\convert.py", line 132, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Python27\lib\site-packages\canmatrix\convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "C:\Python27\lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "C:\Python27\lib\site-packages\canmatrix\formats\__init__.py", line 94, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "C:\Python27\lib\site-packages\canmatrix\formats\arxml.py", line 1723, in load
    frame = get_frame(frameTrig, search_point, multiplex_translation, ns, float_factory)
  File "C:\Python27\lib\site-packages\canmatrix\formats\arxml.py", line 1419, in get_frame
    get_signals(pdu_sig_mapping, new_frame, root_or_cache, ns, None, float_factory)
  File "C:\Python27\lib\site-packages\canmatrix\formats\arxml.py", line 1190, in get_signals
    new_signal.add_values(key, value)
  File "C:\Python27\lib\site-packages\canmatrix\canmatrix.py", line 296, in add_values
    self.values[int(value)] = valueName
  File "C:\Python27\lib\site-packages\future\types\newint.py", line 76, in __new__
    return super(newint, cls).__new__(cls, val)
ValueError: invalid literal for long() with base 10: '0b0000000000000000'

C:\Git\canmatrix>`

Python 3.6
`C:\Git\canmatrix-0.9.1>python examples\convert.py examples\latest.arxml latest.dbc
INFO - convert - Importing examples\latest.arxml ...
Traceback (most recent call last):
  File "examples\convert.py", line 6, in <module>
    canmatrix.cli.convert.cli_convert()
  File "C:\Python36\lib\site-packages\click-7.0-py3.6.egg\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "C:\Python36\lib\site-packages\click-7.0-py3.6.egg\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "C:\Python36\lib\site-packages\click-7.0-py3.6.egg\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Python36\lib\site-packages\click-7.0-py3.6.egg\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "C:\Python36\lib\site-packages\canmatrix\cli\convert.py", line 132, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Python36\lib\site-packages\canmatrix\convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "C:\Python36\lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "C:\Python36\lib\site-packages\canmatrix\formats\__init__.py", line 94, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "C:\Python36\lib\site-packages\canmatrix\formats\arxml.py", line 1723, in load
    frame = get_frame(frameTrig, search_point, multiplex_translation, ns, float_factory)
  File "C:\Python36\lib\site-packages\canmatrix\formats\arxml.py", line 1419, in get_frame
    get_signals(pdu_sig_mapping, new_frame, root_or_cache, ns, None, float_factory)
  File "C:\Python36\lib\site-packages\canmatrix\formats\arxml.py", line 1190, in get_signals
    new_signal.add_values(key, value)
  File "C:\Python36\lib\site-packages\canmatrix\canmatrix.py", line 296, in add_values
    self.values[int(value)] = valueName
ValueError: invalid literal for int() with base 10: '0b0000000000000000'
C:\Git\canmatrix-0.9.1>`

I installed canmatrix as the following steps
`C:\Git\canmatrix>python setup.py install
C:\Git\canmatrix>pip install -r requirements.txt`

Best Regards
Jungwoo Heo

jungwoo-jeff-heo, 2020-03-02:
Dear @ebroecker 

I added exception handling to canmatrix.py and got the dbc correctly

* canmatrix.py's 289 line
`    def add_values(self, value, valueName):
        """
        Add named Value Description to the Signal.

        :param int or str value: signal value (0xFF)
        :param str valueName: Human readable value description ("Init")
        """
        try:
            self.values[int(value)] = valueName
        except ValueError:
		#Assumed to be hex string
            self.values[int(value, 16)] = valueName`

I am not sure whether it is the best way to solve it, anyway I got what I need.
Thanks a lot :)

Best Regards
Jungwoo Heo

ebroecker, 2020-03-02:
Hi @jungwoo-jeff-heo 

could you try the branch again.

with https://github.com/ebroecker/canmatrix/commit/c619584c0b80c71f3d9001a6a9a27dcf33e6bc0c

it'll be a nice implementation - maybe

jungwoo-jeff-heo, 2020-03-03:
Dear @ebroecker 

It works perfectly. Thank you so much.

Best Regards
Jungwoo Heo

commit ff4cb5be (2020-03-02): fix #454
commit 70b23616 (2020-03-03): Arxml limits number formats (#457)

* add number decoding in arxml for different basis
* fix #454
closed 2020-03-03 by ebroecker