ebroecker/canmatrix #458: Wrong start_bit for Motorola signal

danielhrisca, 2020-03-09:
Hello,

there is an issue setting the _start_bit_ for Motorola signals.

[image]

For the above message the _start_bit_ of the _Speed_ signal is set to 53, and _get_startbit_ also returns 53; the signal looks like this:

[image]

Commenting out the code from https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/canmatrix.py#L313 to https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/canmatrix.py#L321 leads to presevring the DBC defined _start_bit_ (50) and the following plot which is the correct output

[image]

MatinF, 2020-03-09:
We'd be very interested in getting this resolved as well

ebroecker, 2020-03-09:
I created
[startbit.dbc.txt](https://github.com/ebroecker/canmatrix/files/4308744/startbit.dbc.txt)
and  following code
```python
db = canmatrix.formats.loadp_flat("startbit.dbc")
speed_signal =  db.frames[0].signal_by_name("Speed")

print(speed_signal.get_startbit(True,True))
``` 
gives me: `58`

does this help?

MatinF, 2020-03-09:
I think the issue here is that the `dbc` file in question is indeed somewhat non-standard - but apparently this DBC has been provided by an OEM for decoding a particular set of signals from a real-life application. 

As such, it would be ideal if the module was robust towards this special case, even if it is somewhat irregular - rather than having to change the specific DBC. The problem is that if somebody uses such a DBC in another case, they may not realize that something is wrong and hence they may draw the wrong conclusions from their data - rather than re-engineer the DBC to fit a more standard format.

Hope that provides some context.

ebroecker, 2020-03-09:
Well - I did not reengineer the dbc but only tried to create an exact copy of your dbc.

could you have a look on it - maybe compare - if I did the dbc copy right?

MatinF, 2020-03-09:
Arh, I'm sorry for being unclear on that Eduard - I've attached a sample of the DBC here:

[example.zip](https://github.com/ebroecker/canmatrix/files/4308812/example.zip)

ebroecker, 2020-03-09:
please try this code:
```python
import canmatrix.formats

db = canmatrix.formats.loadp_flat("example.dbc")
speed_signal =  db.frames[0].signal_by_name("Speed")

print(speed_signal.get_startbit(True,True))
```

I get `58` (and some errors, but they only exist because the dbc was edited by hand)

MatinF, 2020-03-10:
@danielhrisca maybe you're able to clarify on this?

danielhrisca, 2020-03-10:
I've played a bit with the signals and I think there is still a problem; please look at the GPS_Lap_Cents signal start bit values from the command prompt

[image]

danielhrisca, 2020-03-10:
@ebroecker would it make sense to simply add a new attribute to the Signal class that holds the original start_bit value as found in the database, without any processing?

ebroecker, 2020-03-10:
@danielhrisca 
with the holding approach you won't get the displayed starbit either, just look into dbc:
while speed is 58 in the image above it's 50 in dbc.
GPS_Lap_Cents 51 vs 41. (this values are from the former provided dbc and the former provided images not the latest one, there I don't have the dbc)

=> The usual display of dbc in candb++ differes from actual content in dbc.

For the values displayed in candb++ you have to use get_start_bit maybe with different arguments for intel and motorola coded signals.

danielhrisca, 2020-03-10:
> while speed is 58 in the image above it's 50 in dbc.
> GPS_Lap_Cents 51 vs 41. (this values are from the former provided dbc and the former provided images not the latest one, there I don't have the dbc)

The raw value from the DBC would be exactly what I need for my use case. I think it would be a minimal effort to have this stored in the Signal object. 

```
BO_ 1553 GPS_Data_02: 8 GPS_DL
 SG_ Altitude : 0|16@1+ (0.1,0) [0|6553.5] "" Vector__XXX
 SG_ Satellites : 18|6@1+ (0.5,0) [0|31] "" Vector__XXX
 SG_ GPS_Error : 17|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ GPS_Lap : 16|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ SW_revision : 24|8@1+ (1,0) [0|255] "" Vector__XXX
 SG_ GPS_Finishing_Line : 39|1@1+ (1,0) [0|1] "" Vector__XXX
 SG_ GPS_Lap_Cents : 35|7@0+ (1,0) [0|127] "" Vector__XXX
 SG_ Speed : 44|20@0+ (1,0) [0|511] "" Vector__XXX
```
I still can't see how the start bit values for the GPS_Lap_Cents (46, 46, 41, 51 from my last screenshot) relate to the definition above

ebroecker, 2020-03-10:
it's simply like that:

```python
import canmatrix.formats

db = canmatrix.formats.loadp_flat("example.dbc")

for signal in db.frames[0].signals:
    print("{} Startbit: {}".format(signal.name, signal.get_startbit(bit_numbering=1)))
```

gives me:
```
Altitude Startbit: 0
Satellites Startbit: 18
GPS_Error Startbit: 17
GPS_Lap Startbit: 16
SW_revision Startbit: 24
GPS_Finishing_Line Startbit: 39
GPS_Lap_Cents Startbit: 35
Speed Startbit: 44
```

danielhrisca, 2020-03-11:
Thank you Eduard for the patience! This solves the problem for me

closed 2020-03-11 by danielhrisca