ebroecker/canmatrix #460: Cannot find frame by id

raragon71, 2020-03-12:
I have been trying to follow the exampleMerge.py script with no luck. I can load my dbc and create an empty one, but when I try to copy one frame from my dbc to the empty one (either by id or name), with the copy_frame method, I got the following error:

**>>> canmatrix.copy.copy_frame(768, front, merge)
Traceback (most recent call last):
       File "<stdin>", line 1, in <module>
       File "/home/raragon71/.local/lib/python3.6/site-packages/canmatrix/copy.py", line 143, in copy_frame
    frame_list = [source_db.frame_by_id(frame_id)]
  File "/home/raragon71/.local/lib/python3.6/site-packages/canmatrix/canmatrix.py", line 1612, in frame_by_id
    if test.arbitration_id == arbitration_id:
  File "/home/raragon71/.local/lib/python3.6/site-packages/canmatrix/canmatrix.py", line 716, in __eq__
    self.id == other.id
AttributeError: 'int' object has no attribute 'id'**

Am I missing something?

ebroecker, 2020-03-12:
Hi @raragon71 

Thanks for your comment.

... No you don't miss something, the "exampleMerge" is probably out-dated.
I'm gonna fix this.

Meanwhile - what is your usecase?
Did you see the build in functionalities of convert: https://canmatrix.readthedocs.io/en/latest/cli.html#extract-and-merge
Maybe this does already your job?

ebroecker, 2020-03-12:
@raragon71 

just have a look on https://github.com/ebroecker/canmatrix/commit/7e7e987dd3d241b890cac77cbfe7d71b4aae7ee5

eduard

raragon71, 2020-03-12:
Hello @ebroecker , thanks for your prompt response! I will definitively check out the capabilities of canconvert... 

These are some of my use cases:
1. Easily compare two dbc files and get a useful report/output...
2. Generate a new dbc from a two existing ones, selectively picking some frames from one dbc, and copying them to the generated dbc, etc...
3. Rename/change id's on an existing dbc (in an automated way)
4. Generate ARXML file from the generated dbc
5. Support of CAN-FD frames...
6. Combine the ARXML file obtained from the generated dbc with an ethernet frames ARXML (still need to define/understand the real scenario)

ebroecker, 2020-03-12:
Hi @raragon71 ,

you are right, dumping was outdated also.
We are now much more typesafe in canmatrix, so some arguments are not allowed anymore...

 https://github.com/ebroecker/canmatrix/commit/eae81eae0d0e4f330074a139d7a069781031b7ed

fixes dumping also

ebroecker, 2020-03-12:
for your wishes:

commit eae81eae (2020-03-12): fix #460
closed 2020-04-15 by ebroecker