ebroecker/canmatrix #461: new arxml not able to open after dbc-->arxml conversion

canshancai, 2020-03-13:
Dear ebroecker,

I tried your canmatrix tool again to convert dbc-->arxml, but seems it is not able to open the new generated arxml file, like the following pic.

[image]

Any idea from your point view?
Many thanks,
Shaoxiao

ebroecker, 2020-03-13:
which arxml version did you export ? `--arxmlExportVersion`

canshancai, 2020-03-13:
> which arxml version did you export ? `--arxmlExportVersion`

Thanks for your remind. 
I didn't notice the version option for conversion.
Now I set it 4.1.0, finally I can open the arxml file now.

Many thanks.
Shaoxiao

ebroecker, 2020-03-13:
I can't find the option in the documentation, so thanks for your ticket.

closed 2020-03-13 by ebroecker