ebroecker/canmatrix #462: Value table parsing in .sym

bigoulours, 2020-03-17:
Hello,

I discovered canmatrix after finding out that some people I work with use .sym files.
When converting .sym to .dbc I noticed that some value tables aren't exported correctly, for example:
`enum Stat(0="n.i.O",1="i.O.",2="IPS ongoing",3="NotAv")`

I don't know if all .sym files are made the same way (I don't use them), but I figured that changing quote_aware_space_split to quote_aware_comma_split in formats/sym.py (line 363) solved my problem (see attachement)
[sym.py.txt](https://github.com/ebroecker/canmatrix/files/4342376/sym.py.txt)

If you can confirm it works for other .sym files it would be great if you add it to the code.

Thank you and keep up the good work!
Bertrand

ebroecker, 2020-03-17:
Hi @bigoulours 

thanks for your contribution.

@altendky, you are the *sym* guy here?! Do you have a bit time to look into it?
Otherwise I'm gonna have a look in it.

altendky, 2020-03-17:
@bigoulours, could you please share an input file?  If not the whole thing at least a stripped down copy that still shows the issue?  Also, whatever command or code you are running.  Your change seems sensible but I'd like to see the real data etc first.  Thanks.

As an example of one I deal with...
https://github.com/altendky/stlib/blob/fa8c6806ec19fc4cb798636a38359685855652f4/examples/develop/devices/factory/EPC_DG_ID247_FACTORY.sym
```
FormatVersion=5.0 // Do not edit this line!
Title="canmatrix-Export"
{ENUMS}
enum AcNetworkSupport(0="Local", 1="Remote", 2="None", 3="N/A")
enum AccessLevel(0="User", 1="Engineering", 2="Factory")
enum Active(0="Inactive", 1="Active", 2="Error", 3="N/A")
enum Baudrate(0="_125K", 1="_250K", 2="_500K", 3="_1M")
```
```

bigoulours, 2020-03-17:
Hi @altendky, thank you for your reply.
In the file I was given, key/value pairs in value tables (enums) are separated by comma only (no space).
Your example would then be:
`FormatVersion=5.0 // Do not edit this line!
Title="canmatrix-Export"
{ENUMS}
enum AcNetworkSupport(0="Local",1="Remote",2="None",3="N/A")
enum AccessLevel(0="User",1="Engineering",2="Factory")
enum Active(0="Inactive",1="Active",2="Error",3="N/A")
enum Baudrate(0="_125K",1="_250K",2="_500K",3="_1M")`

If you try to convert it to dbc with `canconvert example.sym example.dbc` the value tables are incomplete.
[example.sym.txt](https://github.com/ebroecker/canmatrix/files/4344460/example.sym.txt)

As I understand it, your current script uses space as splitting character whereas the splitting character should probably be a comma. 
 
My proposal in my first post was also incomplete. Line 364 should be edited as well in order to strip spaces from parsed keys/values:
`split = canmatrix.utils.quote_aware_comma_split(temp_array[1])
                    temp_array = [s.rstrip() for s in split]`
[sym.py.txt](https://github.com/ebroecker/canmatrix/files/4344481/sym.py.txt)

This way the script should still work with your files but also mines.

altendky, 2020-03-18:
I think the tests I just pushed to #463 show that the proposed solution doesn't work because `quote_aware_comma_split()` doesn't work the way we might think.  It expects the 'segment' to start and end with "s or it doesn't care about them.  `'"a,b", c"d,e", f'` should become `['a,b', 'c"d,e"', 'f']` for our needs but instead becomes `['a,b', 'c"d', 'e"', 'f']`.  So, we may need to look for another helper lib.  Writing our own parsing code doesn't sound fun to me.  For what it's worth, the existing code also fails with this additional test.

I'm pretty booked up so I can't promise a timely turnaround on this, but we'll see what happens.

bigoulours, 2020-03-18:
I see... my assumption about what `quote_aware_comma_split` does was wrong.
One solution (or rather workaround) would be to add a space when there is none.
It's pretty straightforward with regular expressions though you'd have to import re.
[sym.py.txt](https://github.com/ebroecker/canmatrix/files/4348212/sym.py.txt)

altendky, 2020-03-18:
I don't think regexps are a good tool to handle quoting.  For example, `",123="` should be a valid quoted name for an enumerator that doesn't get modified.  I expect that `1="ab" ,12="cd"` would cause problems as well.  I do consider the behavior of `quote_aware_comma_split()` to itself be a bug.  I think the underlying `csv` implementation is presumably fine for its task but it turns out it wasn't appropriate to apply it to ours.

altendky, 2020-03-30:
@ebroecker and @bigoulours, how about this.  I'm not going to resolve it to my satisfaction soon I don't think so how about I step out of the way, you make a PR that handles the `,` and the `, ` cases and merge that etc but leave this ticket open for a better solution later.  Sorry I kept this hung up in the mean time.  I've just got too many big things in my queue that I want to get finished up.

bigoulours, 2020-03-30:
I'm fine with that.
Here my version of sym.py.

[sym.py.txt](https://github.com/ebroecker/canmatrix/files/4402001/sym.py.txt)

I've commented the added code portions with `# Workaround until issue 463 is solved:`
With this workaround, parsing is still done with `quote_aware_space_split` but a space is added between `,` and the first digit each time the pattern `",\d+=` is found.
That way parsed values are unchanged. The worst that can happen is that a space is added to a key (enum description) whenever it contains `",\d+=`, which should be rare (and even in that case I don't see much of a consequence).

I also made a slight change line 151: I made in enum selection in `enum_dict` case insensitive because the Software of my colleague detects a conflict when trying to import a .sym with enum001 and Enum001. Feel free to revert this change if your software supports it.

Best regards,
Bertrand

Seneda, 2020-04-10:
Hi,
I have been running into issues with .sym files too, so i am trying to add fixes for them. I have written a fix which i believe solves this issue.
Reimplementing the `quote_aware_comma_split `function to split strings on all commas which aren't within `""` quote.
I've also added some tests for the util function and the sym.load functions to check it works.
I think they cover the same type of cases as the test you added in #463 @altendky .

Thanks,
Seneda

PR #470 (2020-04-15): Process the .sym file "Title" field. fixes #469 And Fix Enum dictionary comma splitting 
fixes #469 
fixes #462
commit 5915980e (2020-04-15): Process the .sym file "Title" field. fixes #469 And Fix Enum dictionary comma splitting  (#470)

* Implement processing of the Title property of .sym files.
* Fix CanMatrix.attribute to return default as documented
* Set the default for the "Title" attribute properly
* Reimplement the quote_aware_comma_split function to handle spaces
between fields
* Change the sym parser to use quote_aware_comma_split to split enum value
tables. Fixes #462.
Add test cases to show that enums are read in correctly from .sym files
closed 2020-04-15 by ebroecker