ebroecker/canmatrix #471: copy_ecu_with_frames method copies frames from other ECU's

raragon71, 2020-04-14:
It seems that there is a defect in the 'copy_ecu_with_frames' method, specifically in RX frame copy portions that somehow copies other ECU's as well. 
In my use case, as workaround, I manually remove the 'extra ECU's' by using 'del_ecu' method. This gives me the desired dbc.
This is the code:

_dbm = canmatrix.formats.loadp_flat("MRR360_Database_Front_V1_00.dbc")
dbs = canmatrix.CanMatrix()
"""
extract SODFL from master and copy it to target
"""
canmatrix.copy.copy_ecu_with_frames("SODFL", dbm, dbs)
"""
target dbc gets the ECU (sodfl) with all frames and signals tx/rx but for some reason it also leaves other independent ecu's
a quick analysis points to an issue when copying rx frames and signals. 
workaround (see above) manually delete SODFR node
"""
dbs.del_ecu("SODFR")
canmatrix.formats.dumpp({"":dbs}, "sodfl.dbc")__

ebroecker, 2020-04-15:
Hi  @raragon71 ,

thanks for your report.

Are you sure that this is a problem in copy_ecu_with_frames?

While copying all frames for the ECU, canmatrix looks which ohter ECU sends each frame and adds this other ECU also to the target matrix. 

I think this is ok, but could you provide an example? Maybe discussion is more easy than

raragon71, 2020-04-15:
Hey @ebroecker  thanks for the prompt response... Well, I believe the issue is when the method is looking at the RX frames... This is the scenario:

Original dbc has four ECU's defined, namely SODFL, SODFR, IFV, and External_Tool. SODFL and SODFR transmit a bunch of frames and the receiver nodes are IFV and External_Tool for both transmitting nodes. There is no relationship between SODFL and SODFR other than having common receivers for all their transmitted frames. They both also consume some frames from both IFV and External_Tool but are not the same ID's.

So I want to extract SODFL from the dbc and put it in a target dbc. When using 'copy_ecu_with_frames', the method copies SODFL, IFV and External_Tool as expected, but also includes/copies SODFR but only with the RX frames (no TX frames whatsoever).

I believe the rationale is that, because 'copy_ecu_with_frames' had to copy IFV and External_Tools to the target dbc because they are receivers of SODFL, and because IFV and External_Tool also transmit to SODFR, somehow the logic in the method copies SODFR with only RX frames.... However my expectation, and what I think should be the expected behavior in this case, would be to copy the ECU (SODFL) with only the directly related nodes (receivers) and discard anything else.

Does it make sense? I probably overlooking some other scenarios or special conditions... I can post the actual dbc's if that helps...
Thank you
/RA

ebroecker, 2020-04-15:
@raragon71 oh yes - please post the dbc (you can also "pn" it to my email-adress) 
This would help me a lot to understand and reproduce the issue...

raragon71, 2020-04-15:
Hey Eduard,
I'm sending you the original dbc file to your email (I only removed some custom attributes with project info)... Please try to extract node SODFL with 'copy_ecu_with_frames'.
Thank you
/RA

ebroecker, 2020-04-16:
Thanks for the dbc, 
I think I understand this issue now.

ebroecker, 2020-04-16:
Wow - seems I already support your use case - just try:

```python
import canmatrix
dbm = canmatrix.formats.loadp_flat("MRR360_Mod.dbc")
dbs = canmatrix.CanMatrix()

canmatrix.copy.copy_ecu_with_frames("SODFL", dbm, dbs, rx=False)

canmatrix.formats.dumpp({"":dbs}, "out.dbc")
```

ebroecker, 2020-04-16:
The algorithm of `copy_ecu_with_frames(ECU)` is like:
* iterate over all frames in source_db - look if  `ECU` is transmitter 
* iterate over all signals in frame - look if `ECU` is receiver
 If one of these points is true, copy the frame/pdu to the target_db.

While copying the frame, all receivers are copied also.
Thus a signal which is received by someECUx creates a someECUx in target_db.
BUT: for someECUx the target_db is not complete. 
someECUx than only contains the frames, which are also owned by ECU (other frames are not included in target_db)

So someECUx is only a name in target_db but far from complete copied (hopefully).

I could add some option to remove all receivers from target_db other than ECU.
But what, If you copy more than one ecu to target? 
Some idea to handle the ecus which should appear in target?
Maybe only these which are copied...

ebroecker, 2020-04-16:
could you please try branch  iss471

raragon71, 2020-04-16:
Hi Eduard, sorry for the delay, I just read the emails... I'll give it a
try tonight...  I'm in the middle of a cross country relocation so been a
bit busy... Btw I found another potential issue with the cancompare.. I'll
open up another ticket

ebroecker, 2020-04-17:
... cross country relocation  ... during corona - maybe a great challenge - what country you are?

yes - create another ticket for cancompare...

raragon71, 2020-04-17:
I know... Didn't have other option, I changed jobs in February and started
selling my house... I sold it beginning of March just when we started to
take coronavirus seriously here in the USA... So it caught me in the middle
of the pandemic... It has been in fact very challenging, specially having
small children and homeschooling them at the same time...

But well, hopefully next week we will be in Indianapolis, looking for a new
home... another challenge BTW.

I'll update the ticket and create a new one over the weekend

Thanks, pretty nice Python library by the way!

raragon71, 2020-04-21:
Hi Eduard, I finally had a chance to test the change you made to support my request. It works just fine. My use case is to have a master dbc that contains all ECU's of the entire network or sub-network, based on a number of possible existing vehicle network configurations. From this master dbc, we extract the ECU to be configured for certain application. So, I can get all the different dbc's for all OEM's and configurations, without maintaining multiple dbc's. The master dbc is of course not realizable, since the code generator will create instances of all frames and signals, possibly consuming excessive RAM. So the use case is maintainability and optimization, at least for our platform. Thanks for your help!
/RA

ebroecker, 2020-04-21:
Thanks for you feedback.
I merged the branch to development.

Stay healthy!
And a lot of patience with your kids (we have homeschooling here in Germany also - not always that easy)

closed 2020-04-24 by ebroecker