ebroecker/canmatrix #474: J1939 PGN does not include Extended Data Page (EDP) and Data Page (DP) bits

Seneda, 2020-04-22:
I think the code for parsing J1939 IDs isn't quite up to date. 

In the code for the PGN property of the ArbitrationId class, the PGN is only constructed from the PS and PF fields. However, there are two more bits in the 29 bit ID which contribute to the PGN. I believe that initially they were reserved for future expansions, so are only set in newer PGNs.

I think it is simple enough to fix, so I will add some tests and try to make a Pull Request.

[image]

See these sites for reference.

https://www.kvaser.com/about-can/higher-layer-protocols/j1939-introduction/

https://www.vector.com/int/en/know-how/technologies/protocols/sae-j1939/#c103492

Seneda, 2020-04-22:
I also think the existing test case maybe incorrect : 

```
def test_frame_calc_j1939_id():
    # we have to set all j1939 properties in the __init__ otherwise the setters crash
    frame = canmatrix.canmatrix.Frame()
    frame.source = 0x22
    frame.pgn = 0xAAAA
    frame.priority = 3
    assert frame.arbitration_id.id == 0xcaa0022```

I am fairly sure that the PGN of 0xAAAA should put 0xAAAA where 0xAA00 appears in the expected ID.

ebroecker, 2020-04-23:
Hi @Seneda ,

thank you very much for your comments.
I don't have much know how about j1939.
So pretty shure you are right with your hints.

Looking forward for you PR, let me know if you need help

Seneda, 2020-04-23:
@ebroecker The fix i have made breaks some existing tests. The ones which seem to be just generically about extracting the PGN i have changed. 
But there is one test that still fails which I am slightly less confident of changing, though I expect its a related issue. 
It is test ```test_j1939_decoder()``` 

It is expecting to interpret the CAN ID / Data Coimbination : ```canmatrix.ArbitrationId(id = 0xec0000, extended= True), bytearray([0x20,10,0,1,0xff,0x66,0x1,0])```
as a BAM Message. In the decoder it expects to see 0xECFF but is getting 0xEC00, Im struglling to find definitive information.

So far as I can tell there is some slightly inconsistent usage of the term PGN in all of the online freely available j1939 documentation. I think that in the cases where the PDU-Format (second byte of the 18-bit PGN field) is under 240, the PDU-Specific section (first byte) is interpreted as the Destination address, and then the name PGN is sometimes used to describe an 18bit number with the last 8 bits set to 0. So some sources say that for a BAM message the PGN is 0xCE00 and destination address is 0xFF, but in the general sense the "PGN" should then be 0xCEFF, as the general "PGN" is made of a PDU-F or 0xCE and a destination address of 0xFF. 

I think therefore that the arbitration IDs in the test should be changed to contain 0xFF instead of 0x00.
This fixes the first half of the test for me, but not the final part.

Seneda, 2020-04-23:
@ebroecker I think i was over confident in my earlier comments.
I have modified my fixes to more clearly show what is being done. I think some of the test cases i previously modified were actually correct. If we define the PGN as follows:

PGN is the 18 bits of the CAN as specified below, Except that for PDU Format 1 (ie PDU-F < 240) the PDU-S byte will be masked to 0x00

[image]

PR #475 (2020-04-24): Fix J1939 PGN getter/setter to use full 18bits. Fixes #474
Fixes #474 
Remove the criteria that PF must be >= 240 for PS field to be set as
this seems to be incorrect. (It may be that for some specific case the
PS need to not be used, but it isn't documented here and in general the
PGN should be all 18 bits)

Modifed Tests:

test_canmatrix_get_frame_by_pgn()
seemed to be using made up data and didn't account for the DP and EDP
fields in expected PGN

test_canid_repr()
Modified the expected repr value to include the PS field in the PGN.

test_canid_parse_values()
Modified expected PGN to includ [... cut]
commit 9f0f8c34 (2020-04-24): Fix J1939 PGN getter/setter to use full 18bits. Fixes #474 (#475)

* Fix J1939 PGN getter/setter to use full 18bits. Fixes #474

Remove the criteria that PF must be >= 240 for PS field to be set as
this seems to be incorrect. (It may be that for some specific case the
PS need to not be used, but it isn't documented here and in general the
PGN should be all 18 bits)
Modifed Tests:

test_canmatrix_get_frame_by_pgn()
seemed to be using made up data and didn't accound for the DP and EDP
fields in expected PGN

test_canid_repr()
Modified the expected repr value to include the PS field in the PGN.

 [... cut]
closed 2020-04-24 by ebroecker