ebroecker/canmatrix #483: Problems with finding Signals in ARXML that are in Secured-I-PDU

mikesmitti, 2020-05-26:
Hello,
I have a topic, I am not sure if it is a question or an issue.

I got an arxml-file  with CAN-FD, PDUs and so on.

I use asammdf, which calls canmatrix.
In the arxml.py-file the load function returns a db-object with frames in it.
There are different frames in this object, which is correct.

But in this frames, signals are missing.
I get back signals which are in a "Container-I-PDU" (investigated with Vector Autosar Explorer)
But I do NOT get Signals which are in "Secured-I-PDU"

I am sorry, the arxml is clients property. So a can not send it.

Do I do something wrong? 
Is it an issue?

Best regards

ebroecker, 2020-06-24:
you could try branch arxml_sameip (don't let the branch name confuse you)

mikesmitti, 2020-07-14:
Finally I had time to try it and it seems like it works.

Thanks for your help.

merna-sedki, 2021-07-06:
@mikesmitti  would u please specify how did u make it work ?

merna-sedki, 2021-07-06:
> you could try branch arxml_sameip (don't let the branch name confuse you)

i cannot find branch with this name would u please comment with link

ebroecker, 2021-07-06:
it's merged mainline, so should work out of the box

mikesmitti, 2021-07-07:
I did think, the problem vanished, since I didn´t have it in a year - until today.

I have several files, where in some I don´t find a specific signal which is in an securde I-PDU. As mentioned earlier.
Others from the same series of measurement-files are ok, the signal is there and plotable with asammdf.

Further analysis is going on.

I don´t know what I did last year to make it work. Maybe I just tried other signals, which worked.

About the arxml:
There are two different PDUs which have the SAME signals in it. One is secured, the other is not.
Maybe, that is causing the error.

Infos:
I tried it with canmatrix version 9.2 and 9.3, installed via pip.

mikesmitti, 2021-07-12:
I found out some more information.

The whole signalgroup, with my desired signal, is missing. 
The reason seems to be, that there are 2 PDUs in the same frame at the same time, when the signals are missing. It is always the second PDU, from which the signals are missing.

When looking at a file, where in the very same frame there is only one PDU, then everything seems to be fine.

So it seems to be some kind of multiplexing error.

ebroecker, 2021-07-12:
@mikesmitti 
Do you have any chance to give me some more details about the ARXML-Files?

Maybe you PN me something like OEM, maybe ECU-Name - sometimes I have the chance to have a look into it without having exact your file here.

mikesmitti, 2021-07-13:
I will have a look, but i can not send the file itsself.

I opened an issue here as well: @danielhrisca/asammdf#534

Since sometimes the channel selection is complete and sometimes it is not, i am not yet entirely sure, exactly where the problem occures.

ebroecker, 2022-03-31:
is this issue still there?

mikesmitti, 2022-03-31:
I will take a look tomorrow.

mikesmitti, 2022-04-01:
I checked it with version 0.9.5. but  I will have to take a closer look, since I dont´t find any signal from the arxml right now. That was not the case with 0.9.3. After downgrading, the older behaviour reappeared (one specific signal missing). The dbc works fine

ebroecker, 2022-12-15:
Hi @mikesmitti ,

...
a lot of time went down from my side...

could you provide an example, where this issue occures?

mikesmitti, 2022-12-15:
Hi,
i will close the issue, I dont have much time, to..

Regards.

closed 2022-12-15 by mikesmitti