ebroecker/canmatrix #484: Issues in converting arxml to dbc

starandmoons, 2020-06-03:
Good day.
I'm new to AUTOSAR in general. I have been trying to use canmatrix to convert a sample [arxml file from this site.](https://github.com/eerimoq/cantools/blob/master/tests/files/arxml/system-4.2.arxml)
But when I tried converting this, I got the following error. 

```
C:\Test>python -m canmatrix.cli.convert -vv system-4.2.arxml system-4.2.dbc
INFO - convert - Importing system-4.2.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - 3 frames in arxml...
DEBUG - arxml - 3 can-frame-triggering in arxml...
DEBUG - arxml - 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - 7 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
DEBUG - arxml - Busname: Cluster0
DEBUG - arxml - 3 frames found in arxml
DEBUG - arxml - processing Frame: message1
DEBUG - arxml - message1
DEBUG - arxml - Frame Message1, signal {http://autosar.org/schema/r4.0}I-SIGNAL has no system-signal
Traceback (most recent call last):
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\cli\convert.py", line 136, in <module>
    sys.exit(cli_convert())
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 829, in _call_
    return self.main(*args, **kwargs)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\cli\convert.py", line 131, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 94, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1723, in load
    frame = get_frame(frameTrig, search_point, multiplex_translation, ns, float_factory)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1419, in get_frame
    get_signals(pdu_sig_mapping, new_frame, root_or_cache, ns, None, float_factory)
  File "C:\Users\Main\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1020, in get_signals
    if "SYSTEM-SIGNAL-GROUP" in system_signal.tag:
AttributeError: 'NoneType' object has no attribute 'tag'
```

May I know the reason why I am having this error? Are there specific arxml files that canmatrix only supports in converting?

ebroecker, 2020-06-03:
Hi @starandmoons ,

thanks for reporting this issue.
This ARXML is a bit strange, there are some missing references to system-signals.

I made a patch for dealing with this in branch iss484

could you try?

pip install git+https://github.com/ebroecker/canmatrix.git@iss484

starandmoons, 2020-06-03:
It is now converting the arxml. Thank you.

closed 2020-09-25 by ebroecker