ebroecker/canmatrix #485: Nodes not converted to JSON

fpetry, 2020-06-03:
When converting from *.dbc to *.json using `--jsonExportAll` the receiving nodes are not exported:

~~~
VERSION "created by canmatrix"

NS_ : 

BS_:

BU_: CCL_TEST TEST_ECU

BO_ 4 muxTestFrame: 7 TEST_ECU
 SG_ myMuxer M : 53|3@1+ (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig4 m0 : 25|7@1- (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig3 m0 : 16|9@1+ (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig2 m0 : 15|8@0- (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig1 m0 : 0|8@1- (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig5 m1 : 22|7@1- (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig6 m1 : 32|9@1+ (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig7 m1 : 2|8@0- (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig8 m1 : 0|6@1- (1,0) [0|0] ""  CCL_TEST
 SG_ muxSig9 : 40|8@1- (1,0) [0|0] ""  CCL_TEST

BO_ 3 testFrameFloat: 8 TEST_ECU
 SG_ floatSignal2 : 32|32@1- (1,0) [0|0] ""  CCL_TEST
 SG_ floatSignal1 : 7|32@0- (1,0) [0|0] ""  CCL_TEST
[...]
~~~

becomes

~~~
{
    "enumerations": {},
    "messages": [
        {
            "attributes": {},
            "comment": "",
            "id": 4,
            "is_extended_frame": false,
            "length": 7,
            "name": "muxTestFrame",
            "signals": [
                {
                    "attributes": {},
                    "bit_length": 3,
                    "comment": null,
                    "factor": "1",
                    "is_big_endian": false,
                    "is_float": false,
                    "is_multiplexer": true,
                    "is_signed": false,
                    "max": "0",
                    "min": "0",
                    "multiplex": "Multiplexor",
                    "mux_value": null,
                    "name": "myMuxer",
                    "offset": "0",
                    "start_bit": 53,
                    "values": {}
                },
                {
                    "attributes": {},
                    "bit_length": 7,
                    "comment": null,
                    "factor": "1",
                    "is_big_endian": false,
                    "is_float": false,
                    "is_multiplexer": false,
                    "is_signed": true,
                    "max": "0",
                    "min": "0",
                    "multiplex": 0,
                    "mux_value": 0,
                    "name": "muxSig4",
                    "offset": "0",
                    "start_bit": 25,
                    "values": {}
                },
                [...]
~~~

Accordingly, when converting back to *.dbc the missing nodes get filled with `Vector__XXX` and the node list is empty:

~~~
VERSION "created by canmatrix"

NS_ :

BS_:

BU_: 

BO_ 4 muxTestFrame: 7 Vector__XXX
 SG_ myMuxer M : 53|3@1+ (1,0) [0|0] "" Vector__XXX
 SG_ muxSig4 : 25|7@1- (1,0) [0|0] "" Vector__XXX
 SG_ muxSig3 : 16|9@1+ (1,0) [0|0] "" Vector__XXX
 SG_ muxSig2 : 15|8@0- (1,0) [0|0] "" Vector__XXX
 SG_ muxSig1 : 0|8@1- (1,0) [0|0] "" Vector__XXX
 SG_ muxSig5 : 22|7@1- (1,0) [0|0] "" Vector__XXX
 SG_ muxSig6 : 32|9@1+ (1,0) [0|0] "" Vector__XXX
 SG_ muxSig7 : 2|8@0- (1,0) [0|0] "" Vector__XXX
 SG_ muxSig8 : 0|6@1- (1,0) [0|0] "" Vector__XXX
 SG_ muxSig9 : 40|8@1- (1,0) [0|0] "" Vector__XXX

BO_ 3 testFrameFloat: 8 Vector__XXX
 SG_ floatSignal2 : 32|32@1- (1,0) [0|0] "" Vector__XXX
 SG_ floatSignal1 : 7|32@0- (1,0) [0|0] "" Vector__XXX
[...]
~~~

I would expect the nodes to be converted, and after converting back to dbc, having an identical file to the original *.dbc.

ebroecker, 2020-06-03:
Hi @fpetry 

you are right, json is far from complete there is an old issue about: #136

ebroecker, 2023-01-23:
Hi @fpetry 

... after years here is a little fix:
https://github.com/ebroecker/canmatrix/tree/issue_485

closed 2023-01-25 by ebroecker

ALUE, 2023-02-09:
您的邮件已收到！              祝好             刘毅

PR #683 (2023-01-25): add basic ecu info to json format
fixes #485