ebroecker/canmatrix #488: EDS to DBC

madshadsen, 2020-06-23:
Hi there,

I was wondering if you've considered adding an EDS to DBC converter for use with CANopen EDS files? I think it would be really helpful for many users!

mrfrenzy, 2021-10-18:
This would be a *really* useful feature to have as EDS-files are so widespread with a lot of hardware.
The file specification is well documented and free to download https://www.can-cia.org/can-knowledge/canopen/cia306/#

mtslab, 2022-05-20:
If you guys figure out how to do this I will love to see it. I don't know why companies bother on writing EDS files if the standard is dbc.

mrfrenzy, 2023-07-19:
It is great if you want to look at the possibility of implementing this, I will help any way I can. Here is a basic example.
Bürkert makes a motor valve called 3280. You can send an opening percentage 0-100% via CAN and it will open this much. It will also respond back with how much it is actually open, the current operating status and if there are any problems.

To aid communication they supply an EDS file with a lot of information.
Most CANopen devices support NMT, SDO and PDO.
Network Management are used for device status.
Service Data Objects are used for programming.
Process Data Objects are used for communication and starts transmitting as soon as the device is powered up.

To make the example easy let's start with PDO only.
Opening the EDS file in any EDS viewer we can see what PDOs are supported automatically:
[image]
(this can be changed with SDO configuration, but let's not worry about that to begin with).

Receive PDO 1 is a communication object the slave RECEIVES from our master. It has two parameters: CMDdigital and PVdigital.
Receive PDO 2 is a communication object the slave RECEIVES from our master. It has one parameter: SPDigital
Transmit PDO 1 is a communication object the slave SENDS to our master. It has two parameters: CMD and CMD_Disp
etc.

I have made a dbc file with the most important objects as an example, I attach this, the EDS file and a PDF file describing the contents. As you can see all the data in my dbc file can be derived from the EDS file.

[BUER328X_1_14.eds.zip](https://github.com/ebroecker/canmatrix/files/12099902/BUER328X_1_14.eds.zip)
[MA3280-EDS-file-EU-ML.pdf](https://github.com/ebroecker/canmatrix/files/12099901/MA3280-EDS-file-EU-ML.pdf)
[Burkert 3280 dbc.zip](https://github.com/ebroecker/canmatrix/files/12099903/Burkert.3280.dbc.zip)

ebroecker, 2023-07-20:
Hi @mrfrenzy 

Thank you for these details, this will help a lot!

Next I'll build some basic code example

ebroecker, 2023-07-21:
Hi @mrfrenzy,

I assembled some code.
As far as I understood, the arbitration-ids are dependent from the node_id.

So my code create this dbc with node_id 1:

[MotorValve.dbc.zip](https://github.com/ebroecker/canmatrix/files/12134538/MotorValve.dbc.zip)

AIRicky, 2023-08-22:
> Hi there,
> 
> I was wondering if you've considered adding an EDS to DBC converter for use with CANopen EDS files? I think it would be really helpful for many users!

Hi there, 

Do you know how to convert a DBC file to EDS? 

Best,
Ricky

FerrariX, 2023-10-10:
> https://github.com/ebroecker/canmatrix/tree/eds_support_488

Hi,

I'm getting a fatal error of repository not found.

output from pip install git+https://github.com/ebroecker/canmatrix/tree/eds_support_488

  fatal: repository 'https://github.com/ebroecker/canmatrix/tree/eds_support_488/' not found
  error: subprocess-exited-with-error

I've also tried running 'pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[eds]' but still got error testing the posted 'BUER328X_1_14.eds' file.

Thanks

ebroecker, 2023-10-22:
Hi @FerrariX ,

sorry for the maybe wrong link.
You managed to find a working link for checkout.

I forgot to tell:
you need the module canopen also.
so could you try a `pip install canopen` and try again?

danjhunter, 2023-11-08:
> > Hi there,
> > I was wondering if you've considered adding an EDS to DBC converter for use with CANopen EDS files? I think it would be really helpful for many users!
> 
> Hi there,
> 
> Do you know how to convert a DBC file to EDS?
> 
> Best, Ricky

Is it possible to convert a .DBC file to .EDS file with this branch @ebroecker ?

ebroecker, 2023-11-08:
@danjhunter 

Sorry - only Basic Support to decode EDS - encoding maybe not possible, i think dbc lacks needed information for this task

MatinF, 2025-01-07:
Hi Eduard,

I've been trying this conversion of EDS to DBC on a couple of EDS files (I've sent them via your email), but unfortunately they all fail. The most common error seems to be a missing PDO name - see below the full error trace:

```
Microsoft Windows [Version 10.0.19045.5247]
(c) Microsoft Corporation. All rights reserved.

C:\Users\marti\Downloads\eds-files>canconvert 1115.eds test.dbc
xls is not supported
xlsx is not supported
INFO - convert - Importing 1115.eds ...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Scripts\canconvert.exe\__main__.py", line 7, in <module>
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\cli\convert.py", line 158, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\formats\__init__.py", line 71, in loadp
    return load(fileObject, import_type, key, **options)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\formats\__init__.py", line 90, in load
    dbs[key] = module_instance.load(file_object, **options)  # type: ignore
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\formats\eds.py", line 154, in load
    frame.arbitration_id = canmatrix.ArbitrationId(id=frame_id)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<attrs generated init canmatrix.canmatrix.ArbitrationId>", line 4, in __init__
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\canmatrix.py", line 645, in __attrs_post_init__
    raise ArbitrationIdOutOfRange('ID out of range')
canmatrix.canmatrix.ArbitrationIdOutOfRange: ID out of range

C:\Users\marti\Downloads\eds-files>canconvert HE-927325-0002.eds test.dbc
xls is not supported
xlsx is not supported
INFO - convert - Importing HE-927325-0002.eds ...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Scripts\canconvert.exe\__main__.py", line 7, in <module>
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\cli\convert.py", line 158, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\convert.py", line 68, in convert
[... 63 lines cut ...]
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\cli\convert.py", line 158, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  File "C:\Users\marti\AppData\Local\Programs\Python\Python311\Lib\site-packages\canmatrix\formats\eds.py", line 150, in load
    frame = canmatrix.canmatrix.Frame(name=pdo_name, transmitters=[node_name])
                                           ^^^^^^^^
UnboundLocalError: cannot access local variable 'pdo_name' where it is not associated with a value

```

It would be awesome to get this fully tuned for EDS to DBC conversion and potentially integrated into the canmatrix master - this would help a ton of our users.

ebroecker, 2025-01-08:
Hi,

"UnboundLocalError: cannot access local variable 'pdo_name' where it is not associated with a value" is fixed in branch :
https://github.com/ebroecker/canmatrix/tree/eds_bugfixing

I also tried to implement factor and offset.

MatinF, 2025-01-08:
Thanks Eduard, now this converts the various EDS files I have in my test folder.

Some observations about the outputs:

1) There seems to be a DBC syntax error in most DBCs created when I check via trying to open them via CANDB++, e.g. `sw001454.dbc` from the samples I sent does not open. It seems to be related to some missing entries perhaps in the top section related to the use of sender/receiver nodes later in the DBC

2) There is generally a number of uses of `"` in the DBC due to direct extraction of this from the EDS. Maybe these can get replaced to e.g. `'` or something in order to avoid this, as the character is considered invalid in DBC files

3) There are several cases where there are issues with the sender/receiver nodes. E.g. some use spaces in the name etc, some are missing for some signals and some are missing for some messages - all of these cause the DBC to not open via CANDB++, but I think they are fairly quick to solve. A simple overall solution may be to skip these entirely and just use the `Vector__XXX` for all of these - this way I think a lot of robustness is gained with no loss in terms of data value.

4) There are multiple cases where the names are too long in the EDS, not sure if there's a great solution to this other than adding a somewhat arbitrary replacement/capping rule, so this is more as an observation.

5) Perhaps most importantly it seems like signal names are not getting extracted within the relevant PDOs correctly. Below is an example for the `1115.eds` file where there are multiple PDOs with signals mapped, e.g. below:

[image]

However, the message in that example screenshot seems to contain some default dummy values:

```

BO_ 641 Transmit_PDO_2_Mapping: 8 
 SG_ PDO_Mapping_Entry0 : 0|32@1+ (1,0) [0|4294967295] "" PLC
 SG_ PDO_Mapping_Entry1 : 32|32@1+ (1,0) [0|4294967295] "" PLC
 SG_ PDO_Mapping_Entry2 : 64|32@1+ (1,0) [0|4294967295] "" PLC
 SG_ PDO_Mapping_Entry3 : 96|32@1+ (1,0) [0|4294967295] "" PLC
 SG_ PDO_Mapping_Entry4 : 128|32@1+ (1,0) [0|4294967295] "" PLC
 SG_ PDO_Mapping_Entry5 : 160|32@1+ (1,0) [0|4294967295] "" PLC
 SG_ PDO_Mapping_Entry6 : 192|32@1+ (1,0) [0|4294967295] "" PLC
 SG_ PDO_Mapping_Entry7 : 224|32@1+ (1,0) [0|4294967295] "" PLC
```

ebroecker, 2025-01-08:
Hi @MatinF ,

1. and maybe 2. and 3.  should be fixed

4. long names are no problem, there are some warnings, but it should work out of the box

for 5. I have no solution for now, maybe I am not using the canopen-python

MatinF, 2025-01-09:
Hi Eduard, thanks for the updates.

I have created a small script below that essentially identifies the various mapped parameters linked to the receive/transmit PDOs and which then correctly loads the start bit, bit length, name and data type (required to e.g. determined if signed/unsigned). I think this should be fairly simple to incorporate into your existing EDS code in order to correctly load the mapped signals and link them to the PDOs.

As for the scale/offset aspect, this (unfortunately) seems to be non-trivial. From what I can tell, EDS files do not directly incorporate this information in the same way that DBC files does. As such, some of the EDS files I have provide this detail in "comments" below the parameters, but in a completely non-standardized manner, making it impossible to extract this directly. But users will have a much better starting point for their DBC if we can make these final updates that will allow them to quickly get all the PDOs mapped correctly. 

If you get to update this, I will try out the updated version subsequently.

```
import canopen

# Define the Node ID (default is 0)
node_id = 0

# Load the EDS file into the object dictionary
object_dictionary = canopen.import_od("1115.eds", node_id=node_id)

# CANopen data type mapping
DATA_TYPE_NAMES = {
    0x01: "BOOLEAN",
    0x02: "INTEGER8",
    0x03: "INTEGER16",
    0x04: "INTEGER32",
    0x05: "UNSIGNED8",
    0x06: "UNSIGNED16",
    0x07: "UNSIGNED32",
    0x08: "REAL32",
    0x09: "VISIBLE_STRING",
    0x0A: "OCTET_STRING",
    0x0B: "UNICODE_STRING",
    0x0C: "TIME_OF_DAY",
    0x0D: "TIME_DIFFERENCE",
    0x0F: "DOMAIN",
    0x10: "INTEGER24",
    0x11: "REAL64",
    0x12: "INTEGER40",
    0x13: "INTEGER48",
    0x14: "INTEGER56",
    0x15: "INTEGER64",
    0x16: "UNSIGNED24",
    0x18: "UNSIGNED40",
    0x19: "UNSIGNED48",
    0x1A: "UNSIGNED56",
    0x1B: "UNSIGNED64",
}

def get_data_type_name(data_type_code):
    return DATA_TYPE_NAMES.get(data_type_code, f"Unknown (0x{data_type_code:X})")

def extract_pdos(od, comm_range, map_range, pdo_type):
    print(f"\nExtracting {pdo_type} PDOs...")
    for comm_index, map_index in zip(range(comm_range[0], comm_range[1]), range(map_range[0], map_range[1])):
        if comm_index not in od or map_index not in od:
            continue

        # Retrieve the COB-ID
        comm_param = od[comm_index] #od.get(comm_index)
        cob_id_entry = comm_param.get(1) if comm_param else None
        if not cob_id_entry or cob_id_entry.default is None:
            print(f"  Warning: No valid COB-ID found for {pdo_type} PDO at index 0x{comm_index:04X}. Skipping.")
            continue

        cob_id = cob_id_entry.default & 0x7FF
        print(f"\n{pdo_type} PDO at Communication Index: 0x{comm_index:04X}, COB-ID: {cob_id}")

        # Process Mapping Parameters
        mapping_param = od.get(map_index)
        if not mapping_param:
            print(f"  Warning: No mapping parameter found for {pdo_type} PDO at index 0x{map_index:04X}.")
            continue

        num_entries = mapping_param[0].default if 0 in mapping_param else 0
        print(f"  Number of mapped objects: {num_entries}")

        current_bit_start = 0
        for subindex in range(1, num_entries + 1):
            mapping_entry = mapping_param.get(subindex)
            if not mapping_entry or mapping_entry.default is None:
                print(f"  Warning: Subindex {subindex} missing for mapping parameter at 0x{map_index:04X}.")
                continue

            # Decode the mapping entry
            mapping_value = mapping_entry.default
            obj_index = (mapping_value >> 16) & 0xFFFF
            obj_subindex = (mapping_value >> 8) & 0xFF
            bit_length = mapping_value & 0xFF

            # Fetch the mapped object
            mapped_obj = od.get_variable(obj_index, obj_subindex)
            if not mapped_obj:
                print(f"  Warning: Could not find object at Index: 0x{obj_index:04X}, Subindex: {obj_subindex}.")
                continue

            data_type_name = get_data_type_name(mapped_obj.data_type)
            print(f"  Mapped Object {subindex}:")
            print(f"    Name: {mapped_obj.name.replace(' ', '_')}")
            print(f"    Index: 0x{obj_index:04X}")
            print(f"    Subindex: {obj_subindex}")
            print(f"    Start Bit: {current_bit_start}")
            print(f"    Bit Length: {bit_length}")
            print(f"    Data Type: {data_type_name}")

            # Update the bit start position for the next entry
            current_bit_start += bit_length

# Extract RPDOs
extract_pdos(object_dictionary, (0x1400, 0x1408), (0x1600, 0x1608), "Receive")

# Extract TPDOs
extract_pdos(object_dictionary, (0x1800, 0x1808), (0x1A00, 0x1A08), "Transmit")

```

MatinF, 2025-01-09:
As an additional note:

I think in practice it only makes sense for the EDS loading mechanism in canmatrix to support loading of receive/transmit PDOs. This is the part that will be relevant to load (and possibly convert to e.g. DBC) for all use cases I can envision. 

The other messages that currently get loaded are essentially CANopen services like EMCY, NMT, SYNC etc. These are 'generic' messages that are always in every EDS file and have no unique information per CAN node that is relevant in the context of loading the database like this. 

To some extent, the same goes for the SDOs (Service Data Objects). These have an identical role to OBD2/UDS request/response messages, which also implies that there is no real value to adding the SDO messages in the DBC unless one also completes them with very extensive multiplexing information. 

For example, you can use an SDO request to request data on a specific parameter by adding the relevant 'index' of that parameter in the data bytes. Here, an SDO request can e.g. read data from a CANopen slave by adding the requested data parameter index in bytes 1 to 3 (counting from zero). The slave responds with the requested value and again stores the index in bytes 1 to 3 and the value in bytes 4 to 7. In principle, this could be encoded into the DBC file by using extended multiplexing and essentially adding all the parameters that can be requested and correctly adding their indices as multiplexors. But I think it may be a more complex addition.

In short, for the purpose of getting an initial 'working version' of EDS support, I think adding the PDOs only would be a good way to start. This would also ease a work flow where users might have an EDS for each CANopen node and they want to quickly create a DBC for each of them. Here it would be a problem if each DBC contains the same 'generic' messages, in particular if they contain no real value for the decoding. But open to inputs on this.

MatinF, 2025-01-09:
Another update on this:

For the sake of good order, I tried digging a bit into whether it would be possible to include at least the SDO service related to 'reading data' from different nodes, as I think this will often come into play in practice, along with the PDO based data.

I have a small MF4 log file from our CANedge with some data on which I've tested the below mini DBC file:

```
VERSION ""

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_:

BO_ 1409 SDO_TX: 8 Vector__XXX
 SG_ CMD_CCS M : 5|3@1+ (1,0) [0|0] "" Vector__XXX
 SG_ Index m2M : 8|24@1+ (1,0) [0|0] "" Vector__XXX
 SG_ Motor_temperature m8229 : 32|8@1+ (1,0) [0|0] "degC" Vector__XXX
 SG_ Controller_temperature m73766 : 32|8@1+ (1,0) [0|0] "" Vector__XXX
 SG_ APP_1_Max_Fw_Velocity m1060865 : 32|32@1- (1,0) [0|0] "" Vector__XXX
 SG_ Torque_regulator_actual m139291 : 32|16@1- (1,0) [0|0] "quants" Vector__XXX
 SG_ Throttle_voltage m73840 : 32|16@1- (1,0) [0|0] "mV" Vector__XXX

BA_DEF_  "BusType" STRING ;
BA_DEF_DEF_  "BusType" "CAN";

SG_MUL_VAL_ 1409 Motor_temperature Index 8229-8229;
SG_MUL_VAL_ 1409 Index CMD_CCS 2-2;
SG_MUL_VAL_ 1409 Controller_temperature Index 73766-73766;
SG_MUL_VAL_ 1409 APP_1_Max_Fw_Velocity Index 1060865-1060865;
SG_MUL_VAL_ 1409 Torque_regulator_actual Index 139291-139291;
SG_MUL_VAL_ 1409 Throttle_voltage Index 73840-73840;

```

You can also see the DBC visualized in an editor below:

[image]

The DBC uses the following logic:

[... 79 lines cut ...]
                print(f"  M: {combined_value}")
                print(f"  Subindex (dec): {subobj.subindex}")
                print(f"  Name: {subobj.name.replace(' ', '_')}")
                print(f"  Bit Length: {subobj.bit_length if hasattr(subobj, 'bit_length') else 'N/A'}")
                print(f"  Type: Record")
                print(f"  Data Type: {get_data_type_name(subobj.data_type)}")
                print(f"  Access Type: {subobj.access_type}")
                print(f"  Default Value: {subobj.default}")
        elif isinstance(obj, canopen.objectdictionary.ODArray):
            for subobj in obj.values():
                combined_value = int(f"{subobj.subindex:02X}{obj.index:04X}", 16)
                print(f"\n{format_index(obj.index, subobj.subindex)}")
                print(f"  M: {combined_value}")
                print(f"  Subindex (dec): {subobj.subindex}")
                print(f"  Name: {subobj.name.replace(' ', '_')}")
                print(f"  Bit Length: {subobj.bit_length if hasattr(subobj, 'bit_length') else 'N/A'}")
                print(f"  Type: Array")
                print(f"  Data Type: {get_data_type_name(subobj.data_type)}")
                print(f"  Access Type: {subobj.access_type}")
                print(f"  Default Value: {subobj.default}")

# List all objects in the Object Dictionary
list_all_objects(object_dictionary)

```

ebroecker, 2025-01-14:
Hi @MatinF 

thanks for your information. 
I don't understand everything (yet), especially how to get and convert the needed multiplex infos.
But - YES - canmatrix supports enhanced multiplexing
`is_complex_multiplexed` and `muxer_for_signal` are the frame and signal-values for signals are the attributes we should use. 
So it should be possible.
https://github.com/ebroecker/canmatrix/blob/development/tests/test_dbc.py#L284 could be a starting-point

MatinF, 2025-01-15:
Hi Eduard, thanks for the updates!

I tried the new branch and spotted a few things when trying to convert the file `1115.eds`:

1) There are some cases where symbol replacements are needed, e.g. `-` should become `_` to avoid invalid signal names
2) The addition of the `PLC` as the receive/send node is a bit inconsistent, which causes issues in CANDB++ on consistency and validity of the DBC - I think the `PLC` should ideally be added to each message and each signal to avoid issues
3) Some signals get a bit length of 0, e.g. `Manufacturer_Device_Name`, `Device_ID`, `Drive_manufacturer`, `Store_EDS`
4) I believe that the same multiplexed values should be added for both the SDO messages, i.e. if the Node ID is 1, I would expect both ID `0x581` and `0x601` to have the same signals available to them. However, it seems like they are treated differently and I am not sure exactly why
5) I think the incorporation of the multiplexing signal `sdo_down_CMD` is incorrect, it is currently created as `3|5@1-`, while in my testing it should be `5|3@1+` from what I can tell. This change is required in order for the decoding to work as expected
6) It seems some signals are added in the SDO message as expected, for example `Motor_temperature` is included and is added in the same way as in my test SDO DBC file. But other signals such as `Controller_temperature` are not included. Looking at them in the raw `1115.eds` file I think the reason is that the latter signal is mapped with a sub index different from 0:

```
[2025]
ParameterName=Motor temperature
ObjectType=7
DataType=5
AccessType=ro
PDOMapping=1
;;Unit: °C
;;Description: measured motor temperature

[2026sub1]
ParameterName=Controller temperature
ObjectType=7
DataType=5
AccessType=ro
PDOMapping=1
```

In this case, `Controller_temperature` should be included, but the multiplex value of the 3-byte index should be `73766` as per the test DBC file I created manually. It looks like all signals with a subindex are currently not included in the conversion to the DBC file.

7) Finally, the transmit/receive PDOs remain with the dummy values mapped currently, but I assume this is because your commit only focuses on the SDO aspect and multiplexing handling. 

Overall I think this is looking really good and I expect this will be an excellent contribution to canmatrix! Let me know if I can help with any further testing/feedback.

ebroecker, 2025-01-15:
Hi @MatinF 

Tanks for your patience.
I do just step by step...

Hopefully fixed some of your findings.

MatinF, 2025-01-15:
Hi Eduard, awesome with the quick updates!

1) It looks like this resolves 1), 4), 5) and 6), though I'll have to check in further detail
2) There are still the issues with cases where a message or signal lacks a 'node' (e.g. `PLC`), which causes a complete rejection by CANDB++ to open the DBC file until resolved, i.e. as per 2) before
3) I also think 3) and 7) is the same as before
4) I found an issue where some signals, e.g. `Standard_Error_Field0` to `Standard_Error_Field4` are correctly made unique within the message/signal section, but then it seems the signals are just called `Standard_Error_Field` and therefore duplicated in the multiplexing section, causing a CANDB++ syntax error

ebroecker, 2025-01-15:
Hi @MatinF 

hopefully fixed even more points,
* actively remove empty signals (size = 0)
* fix empty signal receivers 
* Standard_Error_Field0 and so on shoult be fixed also

* 7 is still open, not sure what to to with the pdos

MatinF, 2025-01-15:
Awesome, a few observations:

**1)** The issue with 2) is now almost resolved, though some messages still lack the node name (e.g. the PDOs). It seems all signals have them now

**2)** There is a discrepancy between the SDO upload/download. In the `SDO_download` message you use the `sdo_down_CMD` and `sdo_down_IDX` signals, which are correctly referenced in the multiplexing section as per below:

```
SG_MUL_VAL_ 1537 sdo_down_IDX sdo_down_CMD 2-2;
SG_MUL_VAL_ 1537 Device_Type sdo_down_IDX 4096-4096;
SG_MUL_VAL_ 1537 Error_Register sdo_down_IDX 4097-4097;
...
```
However, for the `SDO_upload` you use the signals `sdo_up_CMD` and `sdo_up_IDX`, yet in the multiplexing section you use `sdo_down_IDX`, which causes the DBC to be invalid:

```
SG_MUL_VAL_ 1409 sdo_up_IDX sdo_up_CMD 2-2;
SG_MUL_VAL_ 1409 Device_Type sdo_down_IDX 4096-4096;
SG_MUL_VAL_ 1409 Error_Register sdo_down_IDX 4097-4097;
SG_MUL_VAL_ 1409 Number_of_Errors sdo_down_IDX 4099-4099;
SG_MUL_VAL_ 1409 Standard_Error_Field0 sdo_down_IDX 69635-69635;
SG_MUL_VAL_ 1409 Standard_Error_Field1 sdo_down_IDX 135171-135171;
...
```

As for the PDO mapping, I will try to explain a bit further:

Each CANopen device is able to leverage up to 4 receive PDOs and 4 transmit PDOs. By default, these PDOs might not be used, meaning that the EDS will just show dummy values for them - corresponding to them being empty. However, in most EDS files I have seen, a couple of the PDOs will typically be 'populated' with references to actual CAN signals (e.g. Motor Temperature, Current etc). To showcase how this looks in an EDS GUI, see below (for the `1115.eds`):

[image]

Basically, this is like specifying the encoding of the data payloads of some of the PDOs via the EDS. The way this is done in practice is through references to the object dictionary index/subindex of the parameter of interest within the EDS. Consider e.g. the `Position_actual_value` signal which is specified as below in the EDS:

```
[6064]
ParameterName=Position_actual_value
ObjectType=7
DataType=4
AccessType=rw
PDOMapping=1
;;Unit: user
```
This is referenced in the Transmit PDO 1 mapping entry of the EDS as below:

```
[1a00]
ParameterName=Transmit PDO 1 Mapping
ObjectType=9
SubNumber=9

[1a00sub0]
ParameterName=Number of entries
ObjectType=7
DataType=5
AccessType=rw
PDOMapping=0
DefaultValue=3

[1a00sub1]
ParameterName=PDO Mapping Entry
ObjectType=7
DataType=7
AccessType=rw
PDOMapping=0
DefaultValue=0x60410010

[1a00sub2]
ParameterName=PDO Mapping Entry
ObjectType=7
DataType=7
AccessType=rw
PDOMapping=0
DefaultValue=0x60640020

[1a00sub3]
ParameterName=PDO Mapping Entry
ObjectType=7
DataType=7
AccessType=rw
PDOMapping=0
DefaultValue=0x60770010

[1a00sub4]
ParameterName=PDO Mapping Entry
ObjectType=7
DataType=7
AccessType=rw
PDOMapping=0
DefaultValue=0x5fff0008
```
Here, the index `6064` is referenced along with the sub index `00` and the data length in bits `20` (i.e. 0x20 = 32 bits). 

The above is more to explain how the logic is within the raw EDS. In the script I created earlier, I show how you can use the canopen Python library to extract all of the mapped signals for each PDO, see my previous post below:
https://github.com/ebroecker/canmatrix/issues/488#issuecomment-2579556009

The addition of these actual mapped PDOs would be very useful.

Let me know if the above makes sense, else I'm happy to clarify further.

MatinF, 2025-01-17:
One other note would be to replace `.` with blank or `_` for signals and messages

MatinF, 2025-01-17:
Also, if you get the chance to attempt with the addition of PDO support, it would be awesome - we have a number of users that are actively hoping to be able to do this via canmatrix for their CANopen EDS files and it would be a huge time saver for them. As mentioned, I am happy to try and provide further details on this if it remains unclear how to do it.

ebroecker, 2025-01-17:
Hi @MatinF 

SDO issues should be done, I did implement some more general character-replacing also, so hopefully less errors during candb++ import

I started integrating your PDO-Code (only tx for now - doing step by step)

MatinF, 2025-01-20:
Thanks a ton Eduard, I've done a quick recap of the outstanding items as I see them:

**1)** The transmit PDOs are currently missing the data length in the message syntax (length is set to `0`) - otherwise these seem to work as intended

**2)** It would be great if the receive PDOs could also be mapped, presumably using a similar method

**3)** After having reviewed the SDO communication more, I suggest the below changes to make this more correct:
- Rename them to `SDO_transmit` (0x580 to 0x5FF) and `SDO_receive` (0x600 to 0x67F)
- Rename the `SDO_down_CMD` and `SDO_up_CMD` so that in `SDO_transmit` the 3-bit signal is called `SCS` (Server Command Specifier) and for the `SDO_receive` it's called `CCS` (Client Command Specifier) 
- For both messages rename the index signal to `IDX`
- For the `SDO_transmit` message, if `SCS = 2` it corresponds to an upload response and the data should be decoded accordingly. This is similar to the current implementation. For the `SDO_receive` if `CCS = 1` it is a download request (with data in the payload) and should be decoded accordingly. This is similar to the current implementation, except that for this message the initial multiplexor should be matched when the value is 1, not 2.

I have sent you a visualization of the SDO request/response communication to make the above more clear.

ebroecker, 2025-01-20:
Hi @MatinF 

thanks again for evaluation.

I tried to fix all but "receive PDOs", this should not be a problem but I want to do some cleanup first.

I have a question for implementation:
Is it possible to have a mapping with gaps (PDOs)? 
I mean could it happen, that there is a gap in the signal mappings?
Because I just count the bits of every mapped signal and place them one after another. This would be wrong if there could be holes in the mapping.

MatinF, 2025-01-20:
I am quite certain that there can't be gaps per se, in case a PDO is intended to have a 'gap', typically a dummy signal should be used I believe - I've seen these used in some of the EDS files. I.e. the EDS may define dummy signals of different standard lengths in order to pad the payload to 8 bytes (although I am not sure that is required per se) - and I expect the same principle would have to be used if there was a need to add a 'gap' in the mapping of a PDO. So I believe your current approach should be OK.

MatinF, 2025-01-23:
Thanks again for all your efforts Eduard!

I tried the latest update, I've listed the remaining issues from what I have spotted - I think it's just this one below:

1) It looks like the `sdo_down_IDX` and `sdo_up_IDX` is still used in the multiplexing section for some of the signals, see below example. These should be renamed to `IDX` after which the DBC can open in CANDB++:

```
SG_MUL_VAL_ 1537 IDX CCS 2-2;
SG_MUL_VAL_ 1537 Dummy0005 sdo_down_IDX 5-5;
SG_MUL_VAL_ 1537 Dummy0006 sdo_down_IDX 6-6;
SG_MUL_VAL_ 1537 Dummy0007 sdo_down_IDX 7-7;
SG_MUL_VAL_ 1537 Device_Type sdo_down_IDX 4096-4096;
SG_MUL_VAL_ 1537 Error_Register sdo_down_IDX 4097-4097;
SG_MUL_VAL_ 1537 Highest_sub_index_supported0 sdo_down_IDX 4118-4118;
SG_MUL_VAL_ 1537 Consumer_Heartbeat_Time_1 sdo_down_IDX 69654-69654;
SG_MUL_VAL_ 1537 Consumer_Heartbeat_Time_2 sdo_down_IDX 135190-135190;
SG_MUL_VAL_ 1537 Consumer_Heartbeat_Time_3 sdo_down_IDX 200726-200726;
SG_MUL_VAL_ 1537 Producer_Heartbeat_Time sdo_down_IDX 4119-4119;
SG_MUL_VAL_ 1537 number_of_entries IDX 4120-4120;
SG_MUL_VAL_ 1537 Vendor_Id IDX 69656-69656;
SG_MUL_VAL_ 1537 Product_Code IDX 135192-135192;

```

ebroecker, 2025-01-23:
@MatinF 

thanks again, I have tried to fix this also now

MatinF, 2025-01-23:
I tried the latest version, but it seems the problem remains - I've sent you the EDS I'm testing with via email, I believe you'll be able to replicate the issue with this.

ebroecker, 2025-01-23:
Hi @MatinF 
you are right, I missed to change one line - now it looks better

MatinF, 2025-01-24:
@ebroecker This is great, now it works as intended and all my test EDS files convert as expected. I think this covers all the EDS files I have had access to now and it means we can basically ensure a quick EDS-to-DBC conversion. I will be featuring this in my next article on our website as well.

It would be awesome if you could add both this branch and the LDF branch to the master branch of canmatrix - I think both contribute massive improvements! :-)

ebroecker, 2025-01-24:
the merges will happen, I just want to find some time to fix some minor other things for the next release

closed 2025-01-24 by ebroecker

MatinF, 2025-02-05:
Hi Eduard I've sent you a mail regarding a small bug in the current EDS to DBC conversion, related to signals in PDOs having the wrong signage in some cases. I've sent you some related example files to showcase this by email. It would be awesome if this could be corrected :-)

ebroecker, 2025-02-07:
Hi @MatinF 

I think I found that issue:
https://github.com/ebroecker/canmatrix/pull/838

commit b1521355 (2025-01-20): renaming signals

minor fixes
#488