ebroecker/canmatrix #489: CAN bus multiframe DBC conversion (e.g. for UDS)

madshadsen, 2020-06-25:
Hi all,

I'm looking to convert multiframe CAN bus data (from UDS request/response flows) into human-readable form.

Below could be an example request/response flow under UDS:

```
7E4   |   03 22 01 01 AA AA AA AA (request frame)
7EC   |  10 3E 62 01 01 FF F7 E7 (initial response from ECU)
7E4   |   30 00 00 00 00 00 00 00 (flow control frame)

7EC   |  21 FF BC 16 AE 43 08 00 (payload)
7EC   |  22 00 00 0F CF 09 08 08 (payload)
7EC   |  23 08 08 09 00 00 0A CE (payload)
7EC   |  24 30 CE 4A 00 00 78 00 (payload)
7EC   |  25 00 15 B2 00 00 12 F4 (payload)
7EC   |  26 00 00 08 40 00 00 06 (payload)
7EC   |  27 D1 00 05 F7 06 09 00 (payload)
7EC   |  28 03 00 00 00 00 03 E8 (payload)
```

Originally my thinking would be to use the 1st byte of each of the payload messages as a multiplexor signal and then encode my DBC file to use that and the CAN ID to decode the data. However, it seems that the same CAN ID may be used as the response CAN ID for different UDS request PIDs - meaning that the CAN ID + 1st byte is not enough to uniquely identify what the payload relates to. 

If so, it seems that multiframe decoding is not possible via the standard DBC methodology on a frame-by-frame basis, but instead the software would have to include some rather cumbersome logic to combine the payloads with the original request message information - which seems illogical.

Any thoughts on how this would be done in practice?

ebroecker, 2020-06-25:
hi @madshadsen 

what you need is a UDS decoder, or to be more precise a ISOTP and UDS decoder.

you can use 
* https://github.com/secdev/scapy
* or https://github.com/linklayer/pyvit

* or write something on your own, like I did once: [uds_decode.tar.gz](https://github.com/ebroecker/canmatrix/files/4833490/uds_decode.tar.gz)

madshadsen, 2020-07-03:
Hi @ebroecker, your own uds_decode.tar.gz code is exactly what I was looking for. Is this an official repository and/or package somewhere? I think it would be a great offering for getting a simple UDS response decoding set up.

Ideally I'd like to combine this with DBC conversion using canmatrix ala your frame conversion functionality. Is this something you've worked with previously? For example, once a response is collected by your decoder in "single frames" with >64 bytes, it should be possible to DBC convert those frames by using multiplexing (triggering based on the first two bytes, which contain information on the UDS PID). 

Thanks again.

ebroecker, 2020-07-18:
Well - no - it's not a official repo for now.... 
I created a repr for it: https://github.com/ebroecker/tp_decoders

hope that helps

closed 2020-09-18 by ebroecker