ebroecker/canmatrix #490: Error loading parsing dbc file

liverpoolilove, 2020-07-08:
example:
        SG_ BCM_BrakeSwitchSt : 3|1@0+(1,0)[0|1] "" Vector__XXX
no space between 3|1@0+ and (1,0)

The error message is as follows:

      error with line no: 31
      b' SG_ BCM_BrakeSwitchSt : 3|1@0+(1,0)[0|1] "" Vector__XXX\r\n'

ebroecker, 2020-07-08:
Thanks for reporting this bug.

gonna have a look on it

ebroecker, 2020-07-11:
could you please try branch iss490 ?

thanks

liverpoolilove, 2020-07-14:
> could you please try branch iss490 ?
> 
> thanks

OK!

ebroecker, 2020-07-18:
@liverpoolilove 

die the branch work for you?

liverpoolilove, 2020-08-10:
@ebroecker 
sorry!  i am very busy recently. I modify line 586 of dbc.py, 
before update:  pattern = r"^SG_ +(\w+) *: *(\d+)\|(\d+)@(\d+)([\+|\-]) +\(([0-9.+\-eE]+), *([0-9.+\-eE]+)\) +\[([0-9.+\-eE]+)\|([0-9.+\-eE]+)\] +\"(.*)\" +(.*)"
after update:  pattern = r"^SG_ +(\w+) *: *(\d+)\|(\d+)@(\d+)([\+|\-])+\s*\(([0-9.+\-eE]+), *([0-9.+\-eE]+)\) +\[([0-9.+\-eE]+)\|([0-9.+\-eE]+)\] +\"(.*)\" +(.*)"
It can run normally with or without spaces！

closed 2020-09-20 by ebroecker