ebroecker/canmatrix #491: Converted exe file issue using pyinstaller

manjushmk, 2020-07-09:
Hello,

First of all: Great work on CANMATRIX......

- I have written a lengthy code to do various analysis on dbc file.
- First I converted the dbc file to csv and did various data checks: things work fine here
- The whole code is running fine
- I converted the file using pyinstaller

The converted file doesn't work. 

- It breaks at :
  File "site-packages\canmatrix-0.9.1-py3.6.egg\canmatrix\convert.py", line 46, in convert
TypeError: 'NoneType' object is not iterable

- For testing purpose i just wrote a simple code

####################################
from canmatrix import convert, canmatrix

convert.convert('D:\Script\1.dbc', 'D:\Script\1.csv')

####################################

- Code runs fine in pycharm or in cmd
- But once i convert into exe using pyinstaller, it doesn't 

Any lead could be helpful

ebroecker, 2020-07-11:
I usually do it this way:

`
pyinstaller.exe --hidden-import=canmatrix.cli.convert --hidden-import=canmatrix.formats.dbc --hidden-import=canmatrix.formats.csv --hidden-import=canmatrix.formats.json --hidden-import=canmatrix.formats.dbc --hidden-import=canmatrix.formats.dbf --hidden-import=canmatrix.formats.fibex --hidden-import=canmatrix.formats.kcd --hidden-import=canmatrix.formats.sym --hidden-import=canmatrix.formats.xls --hidden-import=canmatrix.formats.xls_common --hidden-import=canmatrix.formats.xlsx --hidden-import=canmatrix.formats.yaml --hidden-import=canmatrix.formats.arxml --hidden-import=canmatrix.formats.scapy  --hidden-import=canmatrix.formats.wireshark --onefile canmatrix\examples\convert.py 
`

manjushmk, 2020-07-13:
Hello @ebroecker ,
Worked like charm.... The complete code is working now as single exe. 

Thanks for your quick reply....

Stay rocking...

manjushmk, 2020-07-13:
As the issue is resolved, closing the issue.

closed 2020-07-13 by manjushmk

Alex-hxf, 2021-03-31:
> I usually do it this way:
>
> `pyinstaller.exe --hidden-import=canmatrix.cli.convert --hidden-import=canmatrix.formats.dbc --hidden-import=canmatrix.formats.csv --hidden-import=canmatrix.formats.json [...]

hi @ebroecker ,
I installed using this command convert.exe, then I open the exe appears directly after the flash back a message: Usage:

`Usage: convert.exe [OPTIONS] INFILE OUTFILE
Try 'convert.exe --help' for help.

Error: Missing argument 'INFILE'.`

Can you help me to solve the problem?
Thanks a lot!