ebroecker/canmatrix #492: Canconvert : DBC to CSV issue (Sender receiver data inconsistent)

manjushmk, 2020-07-14:
Hello,
Found the following issue.
I am converting dbc file to csv.

In the converted CSV file, ECU Sender Receiver data for last few ECUs are not correct.
The data runs 'OK' for few rows but after say 100th row the data from next column get shifted to left/
In attached picture for ARR.ecu instead of 's' or 'r', the value 0.00654353 gets populated.

[image]

manjushmk, 2020-07-16:
I got one bug in csv.py file:
In the function "def write_ecu_matrix(ecu_name_list, sig, frame, row, col)"
Line # 95
        else:
            row[col] = Pass

Instead it should be 
        else:
            row[col] = ""

With this I am able to solve this issue.

Could you please confirm this as a bug and above fix as possible solution.  If so please incorporate in the next release.

Also there is inconsistency in the data of : 
Name / Phys. Range ,  Function / Increment Unit ,  Value

The values for these are correct in excel sheet.

manjushmk, 2020-07-17:
Hello,
I had to do few modification in csv.py to remove the bugs.
I have attached the modified python file. Please have a look at it and if the fix found relevant please include in next bug release.
Note: the changes have been made in line with xlsx.py

Change lines:
96
[csv_modified for bugfix-MMK.txt](https://github.com/ebroecker/canmatrix/files/4936361/csv_modified.for.bugfix-MMK.txt)

199...216
236...252

ebroecker, 2020-07-18:
Hi @manjushmk ,

thanks, 
I'm gonna double check and integrate...

ebroecker, 2020-09-25:
Hi @manjushmk ,

thanks again.
I finally found the time to merge mainline.

closed 2020-09-25 by ebroecker

manjushmk, 2020-10-01:
GREAT !!!!.

THANK YOU....