ebroecker/canmatrix #495: arxml to dbc/csv has no output at all with python 3.8.3 (anaconda)

daemonPainter, 2020-08-31:
I have an Anaconda install. Via the Anaconda Prompt I `pip install` the program just fine, I can run the help without issues. I have an arxml file (around 10MB) that I'd like to convert into a DBC or CSV file.

I run canconvert IN.arxml OUT.dbc, get the following input, but no dbc file is created. When provided an empty dbc file, nothing changes.

```
INFO - convert - Importing IN.arxml ...
INFO - convert - done

INFO - convert - Exporting OUT.dbc
INFO - convert - done
```

Am I doing something wrong?

ebroecker, 2020-08-31:
Hi @daemonPainter 

Well, what you do looks good.
Do you have CAN-Clusters in your ARXML?

Could you also try `canconvert -vv IN.arxml OUT.arxml` ?

I guess you can't provide your ARXML anyhow?

daemonPainter, 2020-08-31:
Hi @ebroecker , thanks for the quick response!

Yeah, the arxml itself can't be shared for IP reasons 😢 
I've tried that and got this:

```
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - 0 frames in arxml...
DEBUG - arxml - 0 can-frame-triggering in arxml...
DEBUG - arxml - 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - 1057 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done
```

dunno if it helps, it should be mostly ethernet communication going on in this ECU. How can I check for CAN-Clusters?

ebroecker, 2020-09-01:
Well yes. Canmatrix ist CAN only, nur you can try sameip branch.
There ist some more Bus Support...

closed 2020-09-25 by ebroecker