ebroecker/canmatrix #496: LookupError: unknown error handler name '' in dbf.py

nostar, 2020-09-02:
I get this error if I use --ignoreEncodingErrors or not.

```
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix==0.9.1', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/cli/convert.py", line 131, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/formats/__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/formats/__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix/formats/dbf.py", line 521, in dump
    f.write(out_str.encode(dbf_export_encoding, ignore_encoding_errors))
LookupError: unknown error handler name ''
```
By making the following change to dbf.py:313 I no longer get the error
```
ignore_encoding_errors = options.get("ignoreExportEncodingErrors", "ignore")
```

nostar, 2020-09-02:
Looks like something similar when trying arxml->dbc, but this time changing "" to "ignore" in dbc.py doesn't help:
```
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix==0.9.1', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/cli/convert.py", line 131, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/formats/__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix/formats/__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix/formats/dbc.py", line 448, in dump
    f.write((" " + output_names[frame][signal]).encode(dbc_export_encoding, ignore_encoding_errors))
KeyError: Signal(<signal info removed>)
```

nostar, 2020-09-14:
any word on this?

ebroecker, 2020-09-25:
Hi @nostar ,

could you try branch https://github.com/ebroecker/canmatrix/pull/506 ?

nostar, 2020-09-27:
I think this is fixed since 8 out of 13 DBF files were generated before issue #509 happened.

nostar, 2020-09-28:
In iss509 it works if i use --ignoreEncodingErrors, but same error if I dont use that option:
```
LookupError: unknown error handler name ''"
```

ebroecker, 2020-09-28:
ok, thanks.

could you try branch iss496?

nostar, 2020-09-28:
Using iss496 both DBF and DBC fail without --ignoreEncodingErrors.  Both DBF and DBC succeed with --ignoreEncodingErrors:

DBC
```
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix===0caf5fe.post13-g29a8ef5', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/cli/convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/formats/__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/formats/__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/formats/dbc.py", line 401, in dump
    f.write(create_attribute_string(attrib, "SG_", '%d ' % frame.arbitration_id.to_compound_integer() + name, val,
LookupError: unknown error handler name ''
```
DBF
```
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix===0caf5fe.post13-g29a8ef5', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/cli/convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/formats/__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/formats/__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13_g29a8ef5-py3.8.egg/canmatrix/formats/dbf.py", line 531, in dump
    f.write(out_str.encode(dbf_export_encoding, ignore_encoding_errors))
LookupError: unknown error handler name ''
```

ebroecker, 2020-09-28:
Hi @nostar 
thanks for testing.

I'm a bit clueless, because I cannot reproduce.

is there some special char in the signal name (during dbc-export)?
Maybe your dbc is exported to the point before it fails. Comparing with the working '--ignoreEncodingErrors' you could identify the signal.

nostar, 2020-09-28:
Ok, no partial files were generated to use for compare, but I was able to identify the cause of the issue by hard coding ignore_encoding_errors to 'strict':
```
File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post13.dev0_g29a8ef5-py3.8.egg/canmatrix/formats/dbc.py", line 402, in dump
    f.write(create_attribute_string(attrib, "SG_", '%d ' % frame.arbitration_id.to_compound_integer() + name, val,
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2019' in position 98: ordinal not in range(256)
```
It appears my files are unicode files, and this utf-8 character U2019 is a right single quote.  Why a few are encoded like this and so many others use standard ascii for punctuation is beyond me.  I use --db[cf]fExportEncoding='utf-8' which works without the need for --ignoreEncodingErrors.  I think if I actually used any of the DBC/DBF files that dropped these single quotes, I would have found errors in them.

None of this is a canmatrix issue though.  The issue is the blank string passed to 2nd arg of encode(), which is illegal.  It's strange to me that python only complains about passing an illegal argument to this method if it encounters an error.  Anyway, I think this is a proper fix.  This changes the default to 'strict' instead of ''.  The error message from the strict exception made it immediately clear what the problem was.
```
diff --git a/src/canmatrix/cli/convert.py b/src/canmatrix/cli/convert.py
index 68f6945..e67f29c 100644
--- a/src/canmatrix/cli/convert.py
+++ b/src/canmatrix/cli/convert.py
@@ -130,7 +130,7 @@ def cli_convert(infile, outfile, silent, verbosity, **options):
     if options["ignoreEncodingErrors"]:
         options["ignoreEncodingErrors"] = "ignore"
     else:
-        options["ignoreEncodingErrors"] = ""
+        options["ignoreEncodingErrors"] = "strict"
 
     canmatrix.convert.convert(infile, outfile, **options)
     return 0
```

ebroecker, 2020-09-29:
ok, I merged it, so we can close this issue?

nostar, 2020-09-29:
Yessir

closed 2020-09-29 by ebroecker