ebroecker/canmatrix #497: When comes support for AUTOSAR format version 4.3 ?

Florian6451, 2020-09-11:
Hello,
I need to convert a arxml (AUTOSAR format version 4.3) to dbc-file.
Is support for this version planned ?

Thanks
Flo

ebroecker, 2020-09-11:
This should work out of the box.
Did you try?

Florian6451, 2020-09-11:
I'm new in this topic so correct me if I'm wrong, but when I tried to convert the file some signals were missing in the dbc-file.
Exactly the same signals, when I open the arxml-file in AUTOSAR explorer with version 4.2.

ebroecker, 2020-09-12:
hmm, could you investigate a bit deeper?
If only "some" signals are missing, maybe you have some special conditions.
* are the missing signals secured?
* are frames missing or signals?
* are the frames in containers (by the way - is it CANFD?)

Do you have some more Information? I guess you cannot provide the ARXML, but maybe you can tell me which OEM the ARXML from?

Florian6451, 2020-09-15:
It worked now! I used branch: arxml_sameip.
Thanks anyway

closed 2020-09-15 by Florian6451