ebroecker/canmatrix #498: Convert arxml to arxml or dbc, no Output!

weiluncong, 2020-09-13:
C:\Users\10129>canconvert -vv C:\CWL\Python_Engineer\data_rec_and_par\Blackbox_Eth_B018.arxml C:\CWL\Python_Engineer\data_rec_and_par\cwl.arxml
INFO - convert - Importing C:\CWL\Python_Engineer\data_rec_and_par\Blackbox_Eth_B018.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - 0 frames in arxml...
DEBUG - arxml - 0 can-frame-triggering in arxml...
DEBUG - arxml - 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - 0 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting C:\CWL\Python_Engineer\data_rec_and_par\cwl.arxml ...
INFO - convert - done

C:\Users\10129>canconvert -vv C:\CWL\Python_Engineer\data_rec_and_par\Blackbox_Eth_B018.arxml C:\CWL\Python_Engineer\data_rec_and_par\cwl.dbc
INFO - convert - Importing C:\CWL\Python_Engineer\data_rec_and_par\Blackbox_Eth_B018.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml -  Done

DEBUG - arxml - Build arTree ...
DEBUG - arxml -  Done

DEBUG - arxml - 0 frames in arxml...
DEBUG - arxml - 0 can-frame-triggering in arxml...
DEBUG - arxml - 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - 0 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting C:\CWL\Python_Engineer\data_rec_and_par\cwl.dbc ...
INFO - convert - done

Hi, can you tell me how to solve this issue? Thanks a lot!

weiluncong, 2020-09-13:
I can convert dbc to xlsx......feel so amazing.........

ebroecker, 2020-09-18:
looks like your arxml is ethernet-only. There are no CANs in - right?

ebroecker, 2020-09-27:
please reopen, if any further infos available

closed 2020-09-27 by ebroecker