ebroecker/canmatrix #508: TypeError: exceptions must derive from BaseException

nostar, 2020-09-27:
```
$ canconvert --deleteZeroSignals --ignoreEncodingErrors --cutLongFrames=8  file.arxml file.dbf
INFO - convert - Importing file.arxml
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix===0caf5fe.post11-gc6a24f3', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/cli/convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/__init__.py", line 94, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/arxml.py", line 1846, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/arxml.py", line 1744, in decode_can_helper
    frame = get_frame(frameTrig, ea, multiplex_translation, float_factory)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/arxml.py", line 1446, in get_frame
    get_signals(pdu_sig_mapping, new_frame, ea, None, float_factory)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/arxml.py", line 1017, in get_signals
    isignal_array = ea.find_children_by_path(isignal, "I-SIGNAL")
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/arxml.py", line 175, in find_children_by_path
    children = self.get_children(element, path_elements[-1])
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11_gc6a24f3-py3.8.egg/canmatrix/formats/arxml.py", line 220, in get_children
    raise "use follow_all_ref!"
TypeError: exceptions must derive from BaseException
```
I am able to get past this error by commenting out the raise:
```
            #if len(ret) > 0:
                #raise "use follow_all_ref!"
```

ebroecker, 2020-09-27:
this bug is already fixed in pr #426

https://github.com/ebroecker/canmatrix/blob/62722b23bee69299a277ee475978c7e57069e71c/src/canmatrix/formats/arxml.py#L1033

so - stay tuned

ebroecker, 2020-09-27:
should be fixed in develope

nostar, 2020-09-27:
This one is fixed in latest pull.

closed 2020-09-27 by ebroecker